Blueforcer/awtrix3 #621: [FEATURE REQUEST] Ulanzi TC001 Matrix effect

mymmrac, 2024-10-24:
# Feature Request

It would be great to have the original Ulanzi TC001 Matrix effect present in software if it's possible and doesn't violate any licences. The current Matrix effect is ok, but it doesn't look as good as the stock one.

Maybe there are pre-made notifications/custom apps for that already?

[image]

Blueforcer, 2024-10-24:
Checkout the effect page in the documentation 
https://blueforcer.github.io/awtrix3/#/effects

closed 2024-10-24 by Blueforcer

mymmrac, 2024-10-24:
Seems like their matrix effect is pretty simple, 6 different colors move down somewhat randomly (not sure, maybe there is some movement pattern)

I think I will be able to use the official [instruction manual](https://cdn.shopify.com/s/files/1/0136/3119/3188/files/Ulanzi_TC001_Smart_Pixel_Clock_Instruction_Manual.pdf?v=1705976117) to get color values as it includes matrix effect example

Colors (from top to bottom): #003112, #007536, #00ab4f, #3dbc75, #7cd09f, #bde6cd

I will try to use that, maybe I will have some success with that

[image]

mymmrac, 2024-10-24:
It wasn't that difficult actually, I succeeded and got pretty close to the original effect

[image]

@Blueforcer should I create a PR for this, or you don't want to add/change effect?

Effect definition:

```cpp
{"MatrixAdvanced", MatrixAdvanced, EffectSettings(8, ForestColors_p, false)},
```

Effect code:

```cpp
void MatrixAdvanced(FastLED_NeoMatrix *matrix, int16_t x, int16_t y, EffectSettings *settings)
{
    static uint32_t lastMove = 0;

    // Matrix row colors (bottom to top)
    const int colorsCount = 6;
    static CRGB colors[colorsCount] = {
        CRGB(189, 230, 205),
        CRGB(124, 208, 159),
        CRGB(61, 188, 117),
        CRGB(0, 171, 79),
        CRGB(0, 117, 54),
        CRGB(0, 49, 18),
    };

    // Adjust to manage the frequency of new "drops"
    static uint8_t intensity = 16;

    // Create a static matrix to hold the state of each pixel
    static CRGB ledState[32][8];

    // Create a static to hold the color index of top row, -1 - black
    static int topRowColorIndexes[32];

    static bool topRowColorIndexesInit = true;
    if (topRowColorIndexesInit)
    {
        for (uint16_t i = 0; i < 32; i++)
        {
            topRowColorIndexes[i] = -1;
        }
        topRowColorIndexesInit = false;
    }

    uint8_t baseSpeed = 180;                            // Base value for speed calculation
    uint8_t speed = baseSpeed - (settings->speed * 15); // Adjust speed based on settings

    // Update animation based on speed
    if (millis() - lastMove >= speed)
    {
        lastMove = millis();

        // Shift 8 down and update the ledState array
        for (uint16_t i = 0; i < 32; i++)
        {
            for (uint16_t j = 8 - 1; j > 0; j--)
            {
                ledState[i][j] = ledState[i][j - 1];
            }
        }

        // Update top row and spawn new pixels
        for (uint16_t i = 0; i < 32; i++)
        {
            // Update top row colors
            if (topRowColorIndexes[i] != -1)
            {
                // Set top row pixel color
                ledState[i][0] = colors[topRowColorIndexes[i]];

                // Use next color
                topRowColorIndexes[i]++;
                if (topRowColorIndexes[i] >= colorsCount)
                {
                    topRowColorIndexes[i] = -1;
                }
            }
            else
            {
                // Clear color to black
                ledState[i][0] = CRGB(0, 0, 0);
            }

            // Randomly spawn new pixels
            if (topRowColorIndexes[i] == -1 && random8() < intensity)
            {
                topRowColorIndexes[i] = 0;
            }
        }
    }

    // Always draw the current state
    for (uint16_t i = 0; i < 32; i++)
    {
        for (uint16_t j = 0; j < 8; j++)
        {
            matrix->drawPixel(x + i, y + j, ledState[i][j]);
        }
    }
}
```

mymmrac, 2024-10-24:
The matrix effect that is present right now has only 2 colors (spawnColor and trailColor), this effect has 6 different colors and higher intensity, at least for me mine looks better.

No pressure, you can replace the old one or add a new one (as I did), whatever you think is better