Blueforcer/awtrix3 #622: [BUG] No text scrolling

boubaxxx, 2024-10-27:
# Bug report

## Describe the bug

Push a simple text through MQTT in Home assistant to TC001 : 

{
  "text": "Hello, AWTRIX 3!",
  "rainbow": true,
  "duration": 10,
}

The text is displayed but does not scroll.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ 0.96 ]

## To Reproduce

Push a simple text via MQTT to Ulanzi TC001

## Expected behavior

Scrolling text by default.

Is there a specific setting to make the text scroll? I couldn't find it. TC001 has already been flashed 3 times in 0.96.

Blueforcer, 2024-10-27:
Scrolling is always on by default.
Did you ever used the noScroll Option?
or did you changed the scrolling speed?

boubaxxx, 2024-10-27:
I already tried the scrolling speed option, without success
i didnt used NoScroll Option

boubaxxx, 2024-10-27:
I just tried one more time "scrollSpeed" and set it to 50, text began to scroll then, but without it, text wont scroll

Blueforcer, 2024-10-27:
Then the global scroll speed setting was set to 0. See settings API

boubaxxx, 2024-10-27:
Ok thanks for the answer

boubaxxx, 2024-10-27:
Done, set scrollspeed to any other parameter than 0

closed 2024-10-27 by boubaxxx

Blueforcer, 2024-10-27:
It's 100% by default.