Blueforcer/awtrix3 #623: [BUG] WebGUI - menue broken

EmEiBee, 2024-10-29:
System is alive but WebGUI - menue is broken

The error occurred overnight.
The menu shows repeating menu items with some menu items missing

[image]
( Animated GIF when triggering a forced refresh [SHIFT F5] )

  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

Multiple resets of the watch did not help. 
Reloading or even forcing the browser to reload does not help.
Same issue with all available browser.
The system itself works, is responsive and exchanges data via MQTT without errors.

eku, 2024-10-29:
Duplicate to #549. Next time use the search function.

EmEiBee, 2024-10-29:
> Next time use the search function.

There is no need to explicitly mention the search function. The search function only helps if the double has used exactly the words that you would use to describe the problem. An AI that recognizes what I am looking for, even if the search terms I used are not present in the target, does not yet exist (thank God).

closed 2024-10-29 by Blueforcer