Blueforcer/awtrix3 #624: [BUG] Batterylevel is very unprecise on Ulanci TC001

MeisterQ, 2024-10-29:
The batterylevel is very unprecise. If you charge it up to full, it shows 98% on api,

If you leave it running on battery, it will shut off on 48%.

I know ESPs adc isnt very precise, but is there a way to fix it?

Running on awtrix 0.96 on Ulanzi TC001

Blueforcer, 2024-10-29:
The power management and the battery is also very cheap.

You can tweak it with the min and max values in the dev.json.
You will get the actual raw value from the stats topic

MeisterQ, 2024-10-29:
> The power management and the battery is also very cheap.
> 
> You can tweak it with the min and max values in the dev.json. You will get the actual raw value from the stats topic

Where is the dev.json to find at?

MeisterQ, 2024-10-29:
Thank you. Worked

closed 2024-10-29 by MeisterQ