Blueforcer/awtrix3 #626: [BUG] NoMemory when seindg

TimvdEijnden, 2024-11-01:
# Bug report

## Describe the bug

When sending data using REST for each pixel the information is not drawn and response is `ErrorParsingJson`.
When enabling debug_mode I see this in the logs: `[generateNotification]: NoMemory`.

I made a page with a 32x8 grid where I can draw and send the data to the clock. I'm just sending this drawing.
I generate a json payload, see below.

[image]

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

post this payload to `http://${ip}/api/notify`

```
{"draw":[{"dp":[8,0,"#FF0000"]},{"dp":[9,0,"#FF0000"]},{"dp":[10,0,"#FF0000"]},{"dp":[12,0,"#FF0000"]},{"dp":[13,0,"#FF0000"]},{"dp":[14,0,"#FF0000"]},{"dp":[16,0,"#FF0000"]},{"dp":[17,0,"#FF0000"]},{"dp":[18,0,"#FF0000"]},{"dp":[20,0,"#FF0000"]},{"dp":[21,0,"#FF0000"]},{"dp":[22,0,"#FF0000"]},{"dp":[24,0,"#FF0000"]},{"dp":[25,0,"#FF0000"]},{"dp":[26,0,"#FF0000"]},{"dp":[28,0,"#FF0000"]},{"dp":[29,0,"#FF0000"]},{"dp":[30,0,"#FF0000"]},{"dp":[8,1,"#FF0000"]},{"dp":[12,1,"#FF0000"]},{"dp":[14,1,"#FF0000"]},{"dp":[16,1,"#FF0000"]},{"dp":[18,1,"#FF0000"]},{"dp":[20,1,"#FF0000"]},{"dp":[24,1,"#FF0000"]},{"dp":[28,1,"#FF0000"]},{"dp":[5,2,"#FFFFFF"]},{"dp":[8,2,"#FF0000"]},{"dp":[12,2,"#FF0000"]},{"dp":[14,2,"#FF0000"]},{"dp":[16,2,"#FF0000"]},{"dp":[18,2,"#FF0000"]},{"dp":[20,2,"#FF0000"]},{"dp":[24,2,"#FF0000"]},{"dp":[28,2,"#FF0000"]},{"dp":[3,3,"#006400"]},{"dp":[4,3,"#006400"]},{"dp":[5,3,"#006400"]},{"dp":[8,3,"#FF0000"]},{"dp":[9,3,"#FF0000"]},{"dp":[12,3,"#FF0000"]},{"dp":[14,3,"#FF0000"]},{"dp":[16,3,"#FF0000"]},{"dp":[17,3,"#FF0000"]},{"dp":[18,3,"#FF0000"]},{"dp":[20,3,"#FF0000"]},{"dp":[21,3,"#FF0000"]},{"dp":[24,3,"#FF0000"]},{"dp":[25,3,"#FF0000"]},{"dp":[28,3,"#FF0000"]},{"dp":[29,3,"#FF0000"]},{"dp":[2,4,"#006400"]},{"dp":[3,4,"#006400"]},{"dp":[4,4,"#006400"]},{"dp":[8,4,"#FF0000"]},{"dp":[10,4,"#FF0000"]},{"dp":[12,4,"#FF0000"]},{"dp":[14,4,"#FF0000"]},{"dp":[16,4,"#FF0000"]},{"dp":[18,4,"#FF0000"]},{"dp":[20,4,"#FF0000"]},{"dp":[22,4,"#FF0000"]},{"dp":[24,4,"#FF0000"]},{"dp":[26,4,"#FF0000"]},{"dp":[28,4,"#FF0000"]},{"dp":[30,4,"#FF0000"]},{"dp":[2,5,"#006400"]},{"dp":[3,5,"#006400"]},{"dp":[8,5,"#FF0000"]},{"dp":[10,5,"#FF0000"]},{"dp":[12,5,"#FF0000"]},{"dp":[14,5,"#FF0000"]},{"dp":[16,5,"#FF0000"]},{"dp":[18,5,"#FF0000"]},{"dp":[20,5,"#FF0000"]},{"dp":[22,5,"#FF0000"]},{"dp":[24,5,"#FF0000"]},{"dp":[26,5,"#FF0000"]},{"dp":[28,5,"#FF0000"]},{"dp":[30,5,"#FF0000"]},{"dp":[2,6,"#006400"]},{"dp":[3,6,"#006400"]},{"dp":[8,6,"#FF0000"]},{"dp":[10,6,"#FF0000"]},{"dp":[12,6,"#FF0000"]},{"dp":[14,6,"#FF0000"]},{"dp":[16,6,"#FF0000"]},{"dp":[18,6,"#FF0000"]},{"dp":[20,6,"#FF0000"]},{"dp":[22,6,"#FF0000"]},{"dp":[24,6,"#FF0000"]},{"dp":[26,6,"#FF0000"]},{"dp":[28,6,"#FF0000"]},{"dp":[30,6,"#FF0000"]},{"dp":[8,7,"#FF0000"]},{"dp":[10,7,"#FF0000"]},{"dp":[12,7,"#FF0000"]},{"dp":[13,7,"#FF0000"]},{"dp":[14,7,"#FF0000"]},{"dp":[16,7,"#FF0000"]},{"dp":[18,7,"#FF0000"]},{"dp":[20,7,"#FF0000"]},{"dp":[22,7,"#FF0000"]},{"dp":[24,7,"#FF0000"]},{"dp":[26,7,"#FF0000"]},{"dp":[28,7,"#FF0000"]},{"dp":[30,7,"#FF0000"]}],"duration":10}
```

## Expected behavior

The data handled correctly and the drawing is shown.
It should be possible to color all pixels a different color.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

This seems to be the limit:

```
{"draw":[{"dp":[8,0,"#FF0000"]},{"dp":[9,0,"#FF0000"]},{"dp":[10,0,"#FF0000"]},{"dp":[12,0,"#FF0000"]},{"dp":[13,0,"#FF0000"]},{"dp":[14,0,"#FF0000"]},{"dp":[16,0,"#FF0000"]},{"dp":[17,0,"#FF0000"]},{"dp":[18,0,"#FF0000"]},{"dp":[20,0,"#FF0000"]},{"dp":[21,0,"#FF0000"]},{"dp":[22,0,"#FF0000"]},{"dp":[8,1,"#FF0000"]},{"dp":[12,1,"#FF0000"]},{"dp":[14,1,"#FF0000"]},{"dp":[16,1,"#FF0000"]},{"dp":[18,1,"#FF0000"]},{"dp":[20,1,"#FF0000"]},{"dp":[5,2,"#FFFFFF"]},{"dp":[8,2,"#FF0000"]},{"dp":[12,2,"#FF0000"]},{"dp":[14,2,"#FF0000"]},{"dp":[16,2,"#FF0000"]},{"dp":[18,2,"#FF0000"]},{"dp":[20,2,"#FF0000"]},{"dp":[3,3,"#006400"]},{"dp":[4,3,"#006400"]},{"dp":[5,3,"#006400"]},{"dp":[8,3,"#FF0000"]},{"dp":[9,3,"#FF0000"]},{"dp":[12,3,"#FF0000"]},{"dp":[14,3,"#FF0000"]},{"dp":[16,3,"#FF0000"]},{"dp":[17,3,"#FF0000"]},{"dp":[18,3,"#FF0000"]},{"dp":[20,3,"#FF0000"]},{"dp":[21,3,"#FF0000"]},{"dp":[2,4,"#006400"]},{"dp":[3,4,"#006400"]},{"dp":[4,4,"#006400"]},{"dp":[8,4,"#FF0000"]},{"dp":[10,4,"#FF0000"]},{"dp":[12,4,"#FF0000"]},{"dp":[14,4,"#FF0000"]},{"dp":[16,4,"#FF0000"]},{"dp":[18,4,"#FF0000"]},{"dp":[20,4,"#FF0000"]},{"dp":[22,4,"#FF0000"]},{"dp":[2,5,"#006400"]},{"dp":[3,5,"#006400"]},{"dp":[8,5,"#FF0000"]},{"dp":[10,5,"#FF0000"]},{"dp":[12,5,"#FF0000"]},{"dp":[14,5,"#FF0000"]},{"dp":[16,5,"#FF0000"]},{"dp":[18,5,"#FF0000"]},{"dp":[20,5,"#FF0000"]},{"dp":[22,5,"#FF0000"]},{"dp":[2,6,"#006400"]},{"dp":[3,6,"#006400"]},{"dp":[8,6,"#FF0000"]},{"dp":[10,6,"#FF0000"]},{"dp":[12,6,"#FF0000"]},{"dp":[14,6,"#FF0000"]},{"dp":[16,6,"#FF0000"]},{"dp":[18,6,"#FF0000"]},{"dp":[20,6,"#FF0000"]},{"dp":[8,7,"#FF0000"]},{"dp":[10,7,"#FF0000"]},{"dp":[12,7,"#FF0000"]},{"dp":[13,7,"#FF0000"]},{"dp":[14,7,"#FF0000"]},{"dp":[16,7,"#FF0000"]},{"dp":[18,7,"#FF0000"]},{"dp":[20,7,"#FF0000"]}],"duration":10}
```

Blueforcer, 2024-11-01:
I cannot increase the RAM. 
Use less drawing methods or just create a fullscreen GIF and use it.

TimvdEijnden, 2024-11-01:
I could generate an image, can I then upload it using rest?

TimvdEijnden, 2024-11-01:
@Blueforcer How to use a fullscreen GIF?

Blueforcer, 2024-11-01:
> @Blueforcer How to use a fullscreen GIF?

Like a regular icon, just with a 32x8 resolution. You can use the mobile app to create one.

closed 2024-11-14 by Blueforcer