Blueforcer/awtrix3 #627: [BUG] Custom apps lost after reboot/power off

yglodt, 2024-11-12:
Hi,

I noticed that after a reboot, or after the device has been shut down, my custom apps are lost. They are not shown anymore in the app. After I add them again with Postman, they immediately work again.

Is this a bug or a feature maybe?

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.96 ]

p.s.
Thanks for Awtrix, it's great and always impresses people : )

Blueforcer, 2024-11-12:
that's correct, it's not a bug.

yglodt, 2024-11-12:
Thank you for the clarification!

closed 2024-11-12 by yglodt

garanda21, 2024-12-17:
> that's correct, it's not a bug. Apps are holding in RAM.

Sorry for bringing this topic up again. It’s clear that the apps are in RAM, but is there any way to make them permanent? So that they remain after rebooting the device? I’ve seen a folder called `CUSTOMAPPS` in the files, but it’s always empty