Blueforcer/awtrix3 #633: [FEATURE REQUEST] Add retain flag to MQTT topics

Bigmanekb, 2024-11-24:
To integrate AWTRIX with some smart home systems, it is necessary for topics to have the retain flag enabled so that the smart home server can see all devices connected to the broker when it connects.

It is required to add at least one retained topic (e.g., LWT, status: "on") or any other topic that can be used to determine if AWTRIX is connected to the broker.

eku, 2024-11-25:
> It is required to add at least one retained topic (e.g., LWT, status: "on") or any other topic that can be used to determine if AWTRIX is connected to the broker.

Could you please justify that? The brocker knows which client is currently connected. And AWTRIX regularly sends the status.

Bigmanekb, 2024-11-25:
Imagine a smart home system that is turned off. When it powers on and connects to the broker, it retrieves a list of topics to determine which devices are available on the broker. The system checks which topics correspond to connected devices and either creates them or marks them as active.

If a topic does not have the retain flag enabled, the system will not see anything when it starts up.

The topic is searched for either manually or during the system startup. However, if a device does not have a retained status, the system will not see the topic upon any reconnection to the broker and will mark the device as disconnected.

Blueforcer, 2024-11-25:
you're thinking a bit wrong.
if the topic is set to retained. All clients including the smarthome system will receive the latest messages again on each connect from the Broker, even if awtrix is turned off. 
There is absolutely no reason that the system needs the messages on startup, because it's not the current status. it's an old and therefore wrong status. 
You cannot distinguish whether the message comes from Awtrix or whether the broker has sent an old retained message, even if Awtrix is not online.

Bigmanekb, 2024-11-25:
The determination of whether AWTRIX is online or offline is handled by other processes, such as "if the device has not responded for a certain period."

Some smart home systems cannot work with devices without a retained topic because they need to see that such a device exists on the broker in the first place. Whether the device is online or offline is managed by monitoring systems.

It’s not complicated to add a checkbox in the AWTRIX settings to enable or disable the retain flag.

The system, after determining that such a device exists on the broker, will independently decide based on configured criteria whether the device is accessible or not.

However, first, the smart home must recognize that such a device exists at all. This is the main issue.

Blueforcer, 2024-11-25:
Which smarthome system are we talking about?

Bigmanekb, 2024-11-25:
For example, SprutHub operates precisely on these principles. And AWTRIX is currently gaining popularity with this firmware.

Bigmanekb, 2024-11-25:
This approach to working with MQTT reduces unnecessary load from constantly searching for new topics on the broker and enables the processing of thousands of truly useful requests per minute.

Blueforcer, 2024-11-25:
So if I understand correctly, this relatively unknown piece of hardware/software forgets the mqtt devices after every reboot? Isn't that a problem of SpudHub?

Since I see no point in this function and thousands of users have not missed it before, it will not be implemented for the time being, sorry.
If the request is expressed more often in the future, I am ready to include it in the next update.

Until then, you are of course free to add the functionality for you personally to the firmware.

closed 2024-11-25 by Blueforcer

Bigmanekb, 2024-11-25:
> Isn't that a problem of SpudHub?

This is an issue with the protocol itself. Of course, HTTP API can be used to solve tasks, but since MQTT support exists, in my personal opinion, all of its features should also be supported.

Additionally, the current firmware version requires a lot of manual actions. For example, why not allow downloading an icon via an API command? After all, this action is already supported through the web interface. But no — the user has to upload icons as separate files or through assets.

I understand that this is a DIY project, but some things could be made simpler. For instance, upload a flow into the system, and no further actions are required. The flow would handle downloading the necessary icons itself, making it much easier for the user.

Blueforcer, 2024-11-25:
> Additionally, the current firmware version requires a lot of manual actions. For example, why not allow downloading an icon via an API command? After all, this action is already supported through the web interface. But no — the user has to upload icons as separate files or through assets.
> 

Actually, I'm not someone who likes to justify myself, especially since my hundreds of hours of work are free for you.

The reason is, that it is not Awtrix or the ESP32 that downloads the icons but your browser. AWTRIX has too few free resources to handle this task, so its be moved to the user's external hardware.

Bigmanekb, 2024-11-25:
personally, I bought the app :)

Maybe files can probably be uploaded through the API as well? Purely as speculation.

Bigmanekb, 2024-11-25:
In fact, your solution with firmware updates through the web interface is one of the most convenient I’ve encountered in my experience. For that, I extend my heartfelt thanks.

As for our discussion—there’s no difficulty for me in achieving what I need even with the current implementation. I have extensive experience with smart home systems and tools like NodeRED. However, if we consider regular users without my level of expertise, any manual operations become an additional layer of complexity. This increases the barrier to entry. That’s what I was referring to.

I apologize if my comments came across as a reproach. There was absolutely no criticism intended, only observations from my perspective. You’ve done an outstanding job, and for that, you have my deepest gratitude.

Blueforcer, 2024-11-25:
Youre already able to upload the files via http like:
```bash
curl -X POST "http://TARGET_IP/edit?filename=/ICONS/your_icon_name.png" -F "file=@/path/to/your/icon.png"
```

Bigmanekb, 2024-11-25:
> Thats how flows.blueforcer.de uploads the assets.

This wasn’t in the documentation. Thank you!

Supported formats? PNG, JPG, GIF? Requirements for dimensions and other specifications? The documentation only mentions 8x8 JPG.

Blueforcer, 2024-11-25:
https://blueforcer.github.io/awtrix3/#/icons?id=custom-icons

Bigmanekb, 2024-11-25:
> https://blueforcer.github.io/awtrix3/#/icons?id=custom-icons

Thank you!

Bigmanekb, 2024-11-25:
> ```shell
> curl -X POST "http://TARGET_IP/edit?filename=/ICONS/your_icon_name.png" -F "file=@/path/to/your/icon.png"
> ```

curl -X POST "http://TARGET_IP/edit?filename=/ICONS/some.gif" -F "file=@some.gif"

uploads file into root folder
[image]