Blueforcer/awtrix3 #636: [FEATURE REQUEST] secure CDN URL's in web interface from supply chain attacks

thargor, 2024-11-28:
awtrix3 uses CDNs to load libraries, which is generally speaking a good thing, since it saves precious storage space on the chip and compute time to deliver the scripts.

Unfortunately this opens the door to supply chain attacks, where the attacker gains control over a library and injects malicious code to the browser clients.

Although I would consider the risk profile for awtrix3 quite low, it is not hard to secure against this type of attack by using SRI hashes in the html tag to prevent malicious library changes.

Blueforcer, 2024-12-13:
Isn't that the case with 99% of all websites? They are not really unknown scripts.
But as you rightly said, there is not enough memory to store the scripts.

eku, 2024-12-14:
You can secure your browser against this yourself by installing the [LocalCDN](https://www.localcdn.org/) add-on.

thargor, 2024-12-16:
Adding the SRI hashes would protect all users against malicious changes of scripts and only take a few bytes of space.

If I find some time during the holidays I will create a pull request.

The only thing is I can't find any build instructions for the project. Is there any? Am I looking at the wrong places?

eku, 2024-12-16:
> The only thing is I can't find any build instructions for the project. Is there any?

Use PlatformIO to build from the project configuration in the top directory.