Blueforcer/awtrix3 #637: [BUG] Disabling auto brightness not working

thehijacker, 2024-11-28:
# Bug report

## Describe the bug

My devices is next to a Philips Ambilight TV. When TV is on and LED lights around it are lit the brightness on Awtrix display keeps changing. My solution was to detect when TV is turned on, auto brightness would be disabled so it doesn't change all the time. But this is not working correctly.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

## To Reproduce

When I detect TV is on (smart power plug, power usage check) I send this over MQTT (using Home Assistant):

```
actions:
  - action: mqtt.publish
    metadata: {}
    data:
      topic: awtrix_dnevna/settings
      payload: |-
        {       
          "BRI": 10, 
          "ABRI": false
        }
```

When TV is turned off I enabled automatic brightness with:

```
actions:
  - action: mqtt.publish
    metadata: {}
    data:
      topic: awtrix_dnevna/settings
      payload: |-
        {
          "ABRI": true
        }
```

## Expected behavior

I expect the brightness will be set to fixed value of 10 and automatic brightness would be disabled. But instead the brightness still automatically changes depending on the brightness level of the surrounding LED.

## Screenshots

I can make a video of needed.

## Logs

Not sure how to collect logs.

## Additional context

Add any other context about the problem here.

Blueforcer, 2024-12-23:
I cant reproduce it. Works like it should. Maybe there is a problem with the payload in the HA script.
Please try a MQTT clienbt like [MQTT Explorer](https://mqtt-explorer.com/) for testing purposes

Blueforcer, 2025-01-04:
Any news here?

thehijacker, 2025-01-04:
Sorry was out of home. I will test this via mqtt directly today.

thehijacker, 2025-01-05:
I have disabled automations in HA and tested this manually. It indeed works. Automation to turn it on check when power usage went over 200 W for TV but it jumps up and down even to only 20 W. So I updated automation and seems to work now.

closed 2025-01-05 by thehijacker