Blueforcer/awtrix3 #639: [BUG] Random flickering and color changes (noise)

mymmrac, 2024-11-30:
Hi, is my Ulanzi TC001 broken or I am doing something wrong?

I have reinstalled this firmware multiple times and each time I see random pixels flickering and changing colors, most noticeable when display has red color on it, for me seems like some memory issue, but I don't know how to validate that, also this happens right after start, so I don't think this is something related to memory leaking (RAM seems to have ~138k free bytes).
Note: When I bought it was working fine, this started after 2–3 days or using it

Any ideas why this may happen?

[image]

mymmrac, 2024-11-30:
I suppose this may happen because one pixel in the display is dead (at least seems like dead)

[image]

Blueforcer, 2024-11-30:
That's a hardware issue.

Blueforcer, 2024-11-30:
That's a hardware issue. Try to get a replacement

closed 2024-11-30 by mymmrac

kusold, 2024-12-27:
This thread helped me. I opened my clock and immediately flashed awtrix3 onto it, so I was unsure if it was hardware or firmware.

For future visitors, you can set the background color to white with:

```
curl --request POST \
  --url http://${CLOCK_IP}/api/moodlight \
  --header 'content-type: application/json' \
  --data '{"brightness":150,"color":"#FFFFFF"}'
```

When I send that command repeatedly, I always get white up until the same dead pixel and then after that the colors vary.