Blueforcer/awtrix3 #644: [BUG] Icons not working

sysadminsh, 2024-12-13:
# Bug report

## Describe the bug

There is no possibilty to use web interface to download icons. The example error is: Failed to load resource: the server responded with a status of 404. And indeed this is true, example URL from Network tab after clicking is true 404: https://corsproxy.io/?https://developer.lametric.com/content/apps/icon_thumbs/55381.

Morever after trying to download it manually it is png and even conversion to jpg with Photoshop or just extension change results in blank icon on display. 

## Additional information

- Devices involved:
  - Model: Ulanzi TC001
  - awtrix3 version: v 0.96

## To Reproduce

Try to download any icon.

## Expected behavior

Icons should work.

## Screenshots

If applicable, add screenshots to help explain your problem.

[image]

Blueforcer, 2024-12-13:
duplicate of https://github.com/Blueforcer/awtrix3/issues/642 and https://github.com/Blueforcer/awtrix3/issues/362

closed 2024-12-13 by Blueforcer

sysadminsh, 2024-12-13:
[image]

This is my icon and it is in ICONS/55381.jpg and yes it is 8x8 but after placing it in icons there is a place in display where the icon should but it is not. This also apply to another icons that I'm trying to download.

makavelli20, 2024-12-14:
Same here.Uploaded not recognized and download thru UI doesnt work...

Blueforcer, 2024-12-14:
Check out this firmware version for more feedback:
https://github.com/Blueforcer/awtrix3/issues/362#issuecomment-2069952551

makavelli20, 2024-12-14:
It works now.Firmware installed,json deleted and device restarted.After these steps everything works great.Thank you!

sysadminsh, 2024-12-14:
Yes, now it works as for Lametric icons. However I'm still not able to run my own 8x8 JPG icon created in Photoshop. On file browser icon is normally recognized but on display there is empty black place.

Blueforcer, 2024-12-16:
> The download of the icons is working with this firmware, but i got an yellow pixel on the left bottom, which comes and dissappears. The menu also only responds every third time the buttons are pressed. In addition, the entire system seems a bit laggy. Icons work, but everything else somehow doesn’t anymore.

this means awtrix has issues to connect to your MQTT broker. Since the reconnect doesn't run asynchronous, the software stops during the reconnect and "lags".
But this behavior is the same as the official firmware.

mcbookwood, 2024-12-16:
> this means awtrix has issues to connect to your MQTT broker. Since the reconnect doesn't run asynchronous, the software stops during the reconnect and "lags". But this behavior is the same as the [...]

Thanks for the hint. I'll give it another try.

mcbookwood, 2024-12-16:
Works for me now! Thank you!

kRew94, 2024-12-17:
> Hello, i tried updating the firmware mentioned in the other issue, but i get the error: "Update error: Flash Read Failed". What could be the case with this error? I went via the IP and choose update, selected the firmware.bin and clicked on "Update Firmware".

Worked it out, i did a fresh flash of the original firmware and after that I was able to flash the provided firmware.