Blueforcer/awtrix3 #647: HowTo transfer Icons & co. from one device to an other?

EmEiBee, 2024-12-19:
Today I got my second TC001. It's ready to go.
However, I would like to save myself the trouble of entering all the icons and sounds individually into the new TC001.

Hence my question:
Is there a clever way to bring the two pixel clocks up to the same level in this regard?
Or even better:
Is there an option to declare one pixel clock as the master and to bring about an automatic synchronization?

Blueforcer, 2024-12-19:
Just use the backup function in the Webinterface

EmEiBee, 2024-12-19:
Uhhh... Ty... Haven't think about that ^^

EmEiBee, 2024-12-19:
Ok, that works so far but does not set the other things, such as changing the colors of the date display or similar things

Blueforcer, 2024-12-19:
It just transfers files from the flash. API settings must be executed again.

EmEiBee, 2024-12-19:
Ok... I just assumed that these settings were stored somewhere and could be copied...
It's a shame, but it can't be changed.

Thank you again for your help

EmEiBee, 2024-12-19:
... maybe I'm too stupid for that :(
I'm currently trying to put together a code snippet that I can send to a (new) TC001 using "MQTT Explorer" after I've already adjusted most of it using the backup. But somehow it doesn't work.
What's wrong with it?

pixel2/settings
```
{
  "WDCA": "#af7ac5",
  "WDCI": "#797d7f",
  "BRI": 200,
  "ABRI": true,
  "TIM": true,
  "DAT": true,
  "HUM": false,
  "TEMP": false,
  "BAT": false,
  "VOL": 30
}
```
[image]

The only thing I've achieved is that the automatic brightness no longer works... Great job ^^

closed 2024-12-23 by Blueforcer