Blueforcer/awtrix3 #648: [BUG] Icons are not downloaded (even if feedback is "successfull")

P1Rebo, 2024-12-21:
When trying to download a new Icon I enter the icon ID, press "download", "save configuration" and get the "successfull" feedback message.

[image]

But after I want to display the icon, it doesn't appear.

Also, when I go into the "Files" area, I can't see the icon within the "ICONS" folder.

I'm using Edge browser (same behaviour for Chrome) and the newest Awtrix 3 firmware. Also tried the alternative firmware which seems to solve the download issues (but in my case I don't get the error message).

When connecting to the device to download logs (using the flasher tool) the downloaded txt file is empty.

Thanks for a short hint!

UlanziUser, 2024-12-21:
I'm using two Ulanzi TC001 devices with Awtrix 0.96.
Many thanks for this projekt! 

Since few days the download of icons from Lametric is out of function. The preview is ok, but no download.
Is it possible, that Lametric have change the place of the icons on their webserver?

Also I find no software to make 8x8 pixelgraphics, 
The grafik from "www.pixilart.com" was not displayed on the TC001 and with the gif from the old software animake the TC001 is rebooting, when the icon should display.
 I hope, the problem can fixed.

Blueforcer, 2024-12-21:
They blocked the corsproxy I'm using. Will be fixed in the next update.
In the Awtrix 3 app you can create icons and  animations. You have also access to Awtrix icon database.

P1Rebo, 2024-12-21:
@Blueforcer many thanks!

schlagucke, 2024-12-21:
Hello, I bought the app today. Wonderful work. Thank you very much. 

Can more icons be accessed? There are currently just over 1700 available.

Blueforcer, 2024-12-21:
> Can more icons be accessed? There are currently just over 1700 available. @Blueforcer 

For such a small project I think 1700 its a pretty good value 😅 depends on the users creativity 😁

Blueforcer, 2024-12-21:
@P1Rebo do you have enough free flash space available? Did you flashed the firmware with an ease?

schlagucke, 2024-12-21:
> Hello, I bought the app today. Wonderful work. Thank you very much. 
> 
> Can more icons be accessed? There are currently just over 1700 available. @Blueforcer 

I'm a fan of it, no question. The app is particularly useful for me.

P1Rebo, 2024-12-22:
> @P1Rebo do you have enough free flash space available? Did you flashed the firmware with an ease?

Yes, free flash seems to be enough and the installation was no problem. If you want I can test the latest version too.

UlanziUser, 2024-12-22:
I bought the app today and also many thanks for the app! The integration with the TC001 is very good.
With the app i tried to re-create the icon 96 from lametric-site for me. 
It is a difference between the colors in the app and on display of TC001, i try six different versions until i have the right version for me, but this is no criticism.
In the editor a list of the last used colours as selection were helpful.

Blueforcer, 2024-12-22:
> I bought the app today and also many thanks for the app! The integration with the TC001 is very good.
> With the app i tried to re-create the icon 96 from lametric-site for me.
> It is a difference [...]

The colors are correct. Only low brightness will affect the color because of the cheap LEDs.

poblabs, 2024-12-22:
Is it possible to manually download an icon in the meantime? I know how to upload to awtrix but not sure about the download

P1Rebo, 2024-12-25:
> @P1Rebo do you have enough free flash space available? Did you flashed the firmware with an ease?

I just realized that using the temporary firmware the display comes into restart loops after a while... I will try with the newest firmware.

Thorlaroid, 2025-01-01:
> They blocked the corsproxy I'm using. Will be fixed in the next update. In the Awtrix 3 app you can create icons and animations. You have also access to Awtrix icon database.

It seems that the Icon Database in the iOS App cannot be accessed anymore as well. My App is loading continuously and the app is freezing.

Blueforcer, 2025-01-01:
This has nothing to do with this issue nor this repo. 
It's a server issue currently

closed 2025-01-01 by Blueforcer