Blueforcer/awtrix3 #650: [BUG] Matrix doesn't turn on anymore and the clock disconnects from wlan

BaldFabi, 2024-12-23:
# Bug report

## Describe the bug

I just tested the api and sent a power off via `curl` to the clock.

```bash
curl -v http://192.168.100.10/api/power --data '{"power": false}'
```

The matrix turned off as expected. To activate the matrix again I sent the same request as above but with `true`.
The clock didn't turn on. I restarted the clock via api and hardware and even hardware resetted it.
Now the clock is, after a hardware reboot, for a short amount of time reachable via network but presumably disconnects after the clock is visible for around 100ms.
To describe this more in detail:
1. I restart the clock via the left and right buttons on the clock
2. The clock beeps
3. Awtrix and the ip is shown
4. For around 100ms the clock is shown and the matrix turns off
5. The clock isn't reachable in the network

Even if I sent requests to the clock while it's reachable nothing changes.
I also tried to change the brightness and other settings.

## Additional information

- Devices involved:
  - Model: Ulanzi TC001
  - awtrix3 version: 0.96

While booting I'm able to send a `stats` request and I get the following response:

```json
{"bat":0,"bat_raw":0,"type":0,"lux":0,"ldr_raw":0,"ram":142484,"bri":120,"temp":0,"hum":0,"uptime":16,"wifi_signal":-54,"messages":0,"version":"0.96","indicator1":false,"indicator2":false,"indicator3":false,"app":"","uid":"awtrix_fcdb0c","matrix":true,"ip_address":"192.168.100.18"}
```

I know the bat is 0 but thats actually not true it's charged and the matrix is obviously on/`true` but that's correct for that short moment 😆 

## To Reproduce

Steps to reproduce the behavior:

If it's really reproducable "just" send the request above to the clock.

## Expected behavior

I should be able to turn the matrix/clock on again and it shouldn't be unreachable 😆

BaldFabi, 2024-12-23:
Now I'm even more confused. After plugging the clock in and restarting via the buttons the matrix stays on.
Even though I did exactly that before...

Blueforcer, 2024-12-23:
Maybe just a empty battery? 😅

BaldFabi, 2024-12-23:
Nope because the stats endpoints reports

```json
{"bat":100,"bat_raw":700,"type":0,"lux":0,"ldr_raw":165,"ram":141760,"bri":120,"temp":19,"hum":28,"uptime":46,"wifi_signal":-76,"messages":0,"version":"0.96","indicator1":false,"indicator2":false,"indicator3":false,"app":"Temperature","uid":"awtrix_fcdb0c","matrix":true,"ip_address":"192.168.100.18"}
```

The bat can't be charged after 5 seconds.

Btw. now I just sent this request and the matrix is again off

```bash
curl -v http://192.168.100.18/api/custom --data '{"BRI": 170}'
```

Blueforcer, 2024-12-23:
Don't trust the battery readings after boot.
It's a average value, wich needs some readings before getting accurate.
Furthermore there are different batches of the clock out there, wich uses different battery measurement resistors.

So at first step, charge it until the green led lits at the back. This led is controlled by the power board, not the ESP.

After that read the min and max values again and put it in the dev.json for calibrating.

BaldFabi, 2024-12-23:
I will try it again. But what I initially described happened while the clock was plugged in the last 3 months.

BaldFabi, 2024-12-23:
The matrix is still off the stats endpoint reflects this

```json
{"bat":100,"bat_raw":683,"type":0,"lux":0,"ldr_raw":1,"ram":141776,"bri":120,"temp":21,"hum":27,"uptime":27216,"wifi_signal":-77,"messages":0,"version":"0.96","indicator1":false,"indicator2":false,"indicator3":false,"app":"Temperature","uid":"awtrix_fcdb0c","matrix":false,"ip_address":"192.168.100.18"}
```

I also tried to activate the matrix using the following request

```bash
curl -v http://192.168.100.18/api/power --data '{"power": true}'
```

I receive an OK but the matrix is still off. The stats response stays the same as above

Blueforcer, 2024-12-23:
ok i missunderstood.

Correct curl command is
`curl -v http://192.168.100.18/api/power --header "Content-Type: application/json" --data "{\"power\": true}"`
and
`curl -v http://192.168.100.18/api/power --header "Content-Type: application/json" --data "{\"power\": false}"
`
the arduino jsonparser detects the non escaped json always as false, for whatever reason

BaldFabi, 2024-12-24:
Oh ok, good to know.

After a hardware reboot these two requests work. Thank you.

closed 2024-12-24 by BaldFabi