Blueforcer/awtrix3 #651: @all: Strange effect: Runs 24h+, then loses WLAN -> OFF -> ON -> AP mode

EmEiBee, 2024-12-24:
In the meantime, I have a second PixelClock. It is configured as a twin to the first one: same firmware, same settings, same API settings.
Both receive their data in the same go via MQTT and therefore display identically... So really twins.
The new PixelClock, however, does strange things. After 24 hours or more of operation, it logs itself out of the WLAN or, in rare cases, immediately performs a reset.
It doesn't matter whether it has reset itself or whether I have switched it off and on again using the buttons, it then ends up in AP mode again. If I then scan for the WLAN and re-enter the password, it works again for about a day.
It is noticeable that when it starts up, after displaying the version, it somehow stumbles when the "awetrix" lettering is displayed, which is noticeable in a short, usually two-fold flattening of the display. Only once did it go back into the LAN correctly, all the other times, as described above, into AP mode.

So the question is: Does anyone know this problem or even the cause and maybe a way to solve it?

Meanwhile, I wish all Christians a happy and contemplative Christmas Eve!

Greetz
Micha

Blueforcer, 2024-12-24:
If the second one perform different, then it sounds more like an hardware issue. the firmware is the same..

EmEiBee, 2024-12-24:
... yes, that's my fear too. What irritates me is the fact that at least a day has to pass before this problem occurs. It must be related to something that only runs once a day (from the firmware?) and then causes the thing to crash...
I've now taped up the brightness sensor so that it runs at the lower end of the intensity and am letting it run on battery power... Let's see if that has an effect.

EmEiBee, 2024-12-25:
Just an update:
Runs since yesterday on battery without problems. Just some minutes before, I have plugged in the external PSU due the battery was near empty. 
I let it stay like this at minimum 24h to see what's going on.

closed 2025-03-03 by Blueforcer