Blueforcer/awtrix3 #652: [BUG] Not able to enable snow

julius-d, 2024-12-24:
# Not possible to enable snow effect

Thank you very much for this awsome project.

## Describe the bug

I wanted to enable the nice snow effect  on my Awtrix 3, just like last year.
But I failed to do so.

I created a `/dev.json` file and tried
`  "let_it_snow": true` and also ` "background_effect": "Snow"`
so my current   `/dev.json` is
```json
{
  "let_it_snow": true,
  "background_effect": "Snow"
}
```
I also rebooted multiple times.
But there is no snow animation.

## Additional information

- Devices involved:
  - Model: Ulanzi Smart Pixel Uhr 2882
  - awtrix3 version: AWTRIX 3 **0.97**

Thank you very much and a merry christmas ☃️

Blueforcer, 2024-12-24:
This changed in v0.95 from mid year.
Use the settings API with the OVERLAY key

julius-d, 2024-12-24:
Thank you very much for your help. 

I just discovered that I can enable the snow effect also via the Android app. That is nice!

closed 2024-12-24 by julius-d