Blueforcer/awtrix3 #654: Several settings are displayed twice

jarooij, 2024-12-25:
Several settings are displayed twice or more e.g. Icons, Auth, MQTT. When I press one of the duplicate settings the Restart ESP/Save configuration is displayed.

[image]
[image]

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.96

I cleared cache and did a restart on the Ulanzi without result.

Blueforcer, 2024-12-25:
Remove the donottouch.json and restart awtrix(you need to setup Mqtt and time again, or remove the duplicates in this file

jarooij, 2024-12-26:
Thanx, that did the trick. Unfortunately I'm not able to upload icons

Blueforcer, 2024-12-26:
In need more details

jarooij, 2024-12-26:
When I try to upload an icon from https://developer.lametric.com no confirmation message is displayed and the icon is not downloaded in the icon folder.

Blueforcer, 2024-12-26:
Did you deleted the Json or did you manually removed the duplicated entries?

jarooij, 2024-12-26:
I deleted the entire json.

overrider, 2025-01-02:
Same issue here with menu entries getting shown twice somehow - never touched the DoNotTouch.json. Unfortunately i don't know where to remove it; When i click 'Files' i can choose the file in the left hand menu, but the button for Discard never gets active/clickable.

eku, 2025-01-08:
Looks like there's exactly one condition, when the duplicate entries for `param-boxes` are created in https://github.com/Blueforcer/awtrix3/blob/0988887f56b07fa68adeb0c692a379e5f62befd1/src/ServerManager.cpp#L223-L247
 
Look at https://github.com/Blueforcer/awtrix3/blob/0988887f56b07fa68adeb0c692a379e5f62befd1/lib/webserver/esp-fs-webserver.h#L199-L201 it always creates a new one.