Blueforcer/awtrix3 #655: [BUG]Display error of a blueprint (Smart Home Junkie) since update to 0.79

distanzcheck, 2024-12-26:
## Describe the bug

Hello, first of all I wish you a Merry Christmas. 2 or 3 days ago the AWTRIX update 0.97 for the Ulanzi watch was released. Since then the weather forecast has been displaying strange messages. I have already tried to create a new automation with the weather forecast blueprint, unfortunately with the same result. Enclosed I send you a GIF on which you can see the problem directly. It is the lower 2 rows of white points that are now somehow disturbing. It is the same on all my 5 Ulanzis.

The two rows of white pixels at the bottom have only appeared since the update.

[image]

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ v0.97 ]
Greetings Dirk and a happy new year 2025

Blueforcer, 2024-12-26:
I don't know the blueprint. You should first ask smarthomejunkie or in his discord channel what API functions are used in the forecast blueprint. Maybe he needs to adapt his blueprints for API changes.

smarthomejunkie, 2024-12-27:
I checked this. The error is not in the blueprint as far as I can see. The thing is that when you use the graph option in the payload, and also fill the text field, you get this behavior since 0.97. In version 0.96 the text was show without the graph bars underneath the text.

https://github.com/user-attachments/assets/7210e162-856c-4ca5-bcda-f9dea91a217a

Blueforcer, 2024-12-27:
> @Blueforcer: I checked this. The error is not in the blueprint as far as I can see. The thing is that when you use the graph option in the payload, and also fill the text field, you get this behavior since 0.97. In version 0.96 the text was show without the graph bars underneath the text. It happens when you choose the type bar chart. When you choose for line chart, you see a flat line first, but it gets pushed down far enough so that you do not see it in the end.
> 
>  2024-12-27.10-52-34.mp4

I didn't know that the bluetprint uses graph, but how does that make sense if you only have 2 pixels of space under the text?

Blueforcer, 2024-12-27:
@smarthomejunkie can you send me a example json of the weather blueprint output? currently im unable to reproduce it.

smarthomejunkie, 2024-12-27:
It's not the weather app, but the rain forecast app. The TS referred to the wrong blueprint.

The input that I am sending is:

```
    text_case: '0'
    awtrix_display: 9de38def74aa7f10f8aa6af38a8805ad
    toggle_helper: input_boolean.awtrix_display_rain_forecast
    my_sensor: weather.huis
    graph_type: bar
    show_icon: false
    text_color:
      - 0
      - 251
      - 255
    graph_color:
      - 0
      - 179
      - 255
    autoscale: true
    show_rainbow: true
    my_text: No rain expected
    background_color:
      - 0
      - 0
      - 0
    switch_to_app: false
    lifetime: '300'
    awtrix_displays:
      - 5de7ce048caff23cef73a6f2ccb9dd79
      - 92b0eb2aa7b868cd3c4e5e17724bedfe
```

Blueforcer, 2024-12-27:
Please check this 
[firmware.zip](https://github.com/user-attachments/files/18259955/firmware.zip)

smarthomejunkie, 2024-12-27:
Yes, this firmware fixed it! Thank you.

Blueforcer, 2024-12-27:
will be released soon!

distanzcheck, 2024-12-27:
Thank you Blueforcer and SmartHomeJunkie, it works again. There was something good about the message.