Blueforcer/awtrix3 #658: [BUG] MQTT broker does not accepts URL

littlezaphod, 2024-12-31:
# Bug report

## Describe the bug

The MQTT broker requires to be specified as an IP address and cannot be specified as a URL (e.g. mqtt.example.com)

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.96

## To Reproduce

Go through setup process. Enter a valid local MQTT IP address and verify that MQTT connection is established. Now replace IP address with external URL / subdomain of MQTT server, e.g. mqtt.example.com 
In my case I run a reverse proxy and forward mqtt.example.com to local MQTT instance and port.

## Expected behavior

Clock should connect to MQTT. Instead it shows the yellow bottom left pixel indicating MQTT issue.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Jeandhom, 2024-12-31:
URL like mqtt.example.com work fine for me.

Blueforcer, 2024-12-31:
also works with the hostname of my hosted MQTT broker. seems like an issue on your side

littlezaphod, 2024-12-31:
I do have some issues to get my MQTT broker to accept websocket connections which seems to be the issue. You guy are right, problem on my side, issue can be closed

closed 2024-12-31 by littlezaphod