Blueforcer/awtrix3 #663: [BUG] Cant flash second time

smogstate, 2025-01-04:
Hi,

i need some guidance on following:

i have flashed the clock, everything was fine, then i was misconfigured network, so i dont have access to the clock over wifi.
i decided to erase the clock and flash again, was not working showing following error:

[image]

then i pressed small button on the back with a nidle. Now clock showing "AP Mode" and i cant flash it.

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: recent one, couple weeks ago was flashed first time

smogstate, 2025-01-04:
ok its reseted, but i still cant flash to newer version

Blueforcer, 2025-01-04:
There is no boot button on the Ulanzi.

Try another USB cable and USB Port.

smogstate, 2025-01-04:
Other question, i can connect now to web ui in browser everything is fine, but app is still cant discover the clock. i tried to enter ip directly its not working either. how can i troubleshoot this issue?

Blueforcer, 2025-01-04:
Make sure you're on the same Subnet and your router doesn't block broadcast packages.

smogstate, 2025-01-04:
Thanks! Will do, and great work guys! :)

Blueforcer, 2025-01-04:
Im solo, but thanks 🙏

closed 2025-01-05 by Blueforcer