Blueforcer/awtrix3 #665: [BUG] In DNS hostname is not allowed to have an _ in the name

patrick-reijnen, 2025-01-05:
# Bug report

Just found out the my Ulanzi Clock is promoting itself to DNS with hostname **awtrix_116ae8**, resulting in my DNS server complaining about this hostname as shown below. Underscores in hostnames are not allowed in DNS configuration. 

Jan  3 13:54:40 <my DNS server> dhcpd[29755]: DHCPREQUEST for 192.168.1.45 from 48:e7:29:11:6a:e8 (awtrix_116ae8) via enp2s0
Jan  3 13:54:40 <my DNS server> dhcpd[29755]: DHCPACK on 192.168.1.45 to 48:e7:29:11:6a:e8 (awtrix_116ae8) via enp2s0
**Jan  3 13:54:40 <my DNS server> named[3121]: error: zone <my DNS zone name>/IN: awtrix_116ae8.<my DNS zone name>/A: bad owner name (check-names)**
Jan  3 13:54:40 <my DNS server> dhcpd[29755]: Unable to add forward map from awtrix_116ae8.<my DNS zone name>. to 192.168.1.45: REFUSED
Dec 31 01:54:38 <my DNS server> dhcpd[29755]: DHCPREQUEST for 192.168.1.45 from 48:e7:29:11:6a:e8 (awtrix_116ae8) via enp2s0
Dec 31 01:54:38 <my DNS server> dhcpd[29755]: DHCPACK on 192.168.1.45 to 48:e7:29:11:6a:e8 (awtrix_116ae8) via enp2s0
**Dec 31 01:54:38 <my DNS server> named[3121]: error: zone <my DNS zone name>/IN: awtrix_116ae8.<my DNS zone name>/A: bad owner name (check-names)**
Dec 31 01:54:38 <my DNS server> dhcpd[29755]: Unable to add forward map from awtrix_116ae8.<my DNS zone name>. to 192.168.1.45: REFUSED

To be complete, I've added my Ulanzi clock to my Wifi SSID and have my DHCP and DNS server deal with giving out an IP-address and handle DNS names. No static network configuration used.

A simple solution is changing the underscore (not allowed in DNS) by the minus sign (allowed in DNS).

My temporary solution, as I'm maintaining my own DHCP server, is to have my DHCP server rewrite the hostname received from Awtrix (awtrix_116ae8) to awtrix-116ae8 which is accepted by the DNS server.
 
Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: Discovered the issue in version 0.96, and after upgrade also in 0.98. I guess it might be in most / all versions <0.96 too

## To Reproduce

Add a new Ulanzi clock with Awtrix 3 to your network using DHCP (no static network configuration) and watch, if you have access to it, the logs of your DNS server.
Check your logs for DHCP IP renewal requests of the Ulanzi clock and find the DNS requests resulting from the renewal request. 

## Expected behavior

I expect the hostname presented by Awtrix 3 to be accepted as a valid hostname (ie. a hostname with valid characters only) by my DNS server and added to A and PTR records

## Screenshots

NA. 

## Logs

```See the initial bug description```

## Additional context

NA.

Blueforcer, 2025-01-05:
You can change the hostname to fit your needs
https://blueforcer.github.io/awtrix3/#/dev

patrick-reijnen, 2025-01-05:
Thanks, missed that one.

But it would be great if the default name given out by Awtrix 3 is a valid name at least.

closed 2025-03-03 by Blueforcer