Blueforcer/awtrix3 #669: [BUG] HTTP API switch to custom app does not work

gobytrie, 2025-01-06:
# Bug report

## Describe the bug

I create a custom app and send respective updates to my awtrix device.
Whenever I send an Update I also send a switch command via http to the device in order to display the custom app on the device.
Creation and update of custom app works fine, but the switch command does not work. I do receive -> HTTP ERROR 500

I send the following command for custom App.

`http://{ip}/api/custom?name=testapp`

Payload:

`{
    "name":"testapp",
    "text":"test",
}`

And afterwards I want to switch with the following HTTP POST:

`http://{ip}/api/switch?name=testapp`

here I get the HTTP 500 ERROR

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ v0.98 ]

## To Reproduce

Send the above commands via CURL to awtrix device

I also tried other combinations and http requests, but nothing seems to work.

Blueforcer, 2025-01-06:
You have some errors
1. The key "name" in your customapp doesn't exist.
2. To switch to a specific app you need to send a Json body not parameters

Please have a look at the documentation

gobytrie, 2025-01-06:
Ah thank you.
When sending the name of the app as Json body the switch does work correctly.

Currently the documentation states to send the app name as HTTP parameter.

> For custom apps, employ the name you designated in the topic or HTTP parameter. In MQTT, if [PREFIX]/custom/test is your topic, then test would be the app's name.

Could you update the documentation to use the name in the json body and maybe add an example for http post request as there is only one for mqtt.

gobytrie, 2025-01-06:
I find that statement rather confusing. Needed to read it a couple of more times until I got what you meant.
You could rewrite it to something like "to switch to a custom app you need to deploy that name first as http parameter of your custom app".

But maybe that's just me - thanks anyway for clarification on that matter.
I'm going to close this as solved now.

closed 2025-01-06 by gobytrie

Blueforcer, 2025-01-06:
But that would be incorrect. You cannot switch to an app wich doesn't exist.
You only need the name parameter for creating the customapp