Blueforcer/awtrix3 #671: [BUG] API call to settings VOL does not change volume

jasonrichardsmith, 2025-01-07:
# Bug report

## Describe the bug

API call to set volume does not seem to work.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.45 ]

## To Reproduce

Steps to reproduce the behavior:

curl -v http://IP_ADDRESS/api/settings -H "Content-Type: application/json" -d "{'VOL':10}"
curl -v http://IP_ADDRESS/api/settings -H "Content-Type: application/json" -d "{'VOL':30}"
curl -v http://IP_ADDRESS/api/settings -H "Content-Type: application/json" -d "{'VOL':0}"

volume remains the same.

## Expected behavior

volume changes

## Logs

```text
curl http://IP_ADDRESS/api/settings -H "Content-Type: application/json" -d "{'VOL':10}" 
OK%
```

## Additional context

Blueforcer, 2025-01-07:
you need to activate it for the buzzer first. 
https://blueforcer.github.io/awtrix3/#/dev

But i do not recommend it, it doesnt work for many 
tones because you have to keep in mind that it is a PWM volume control. So the base frequency stays the same, but the higher frequency parts change if the volume is changed.

jasonrichardsmith, 2025-01-07:
OK thanks. I will close.