Blueforcer/awtrix3 #674: [BUG] Scrolling longer text (like NINA warnings) "stutters" enormously, making warnings hard to read

Moonbase59, 2025-01-09:
# Bug report

## Describe the bug

Scrolling longer texts—like NINA Warnings—"stutters" a lot, making warnings uncomfortable to read. I _think_ this went smoother on v0.96, but ain't entirely sure. The enclosed GIF recording from Live View doesn't show the stuttering very well, it is much worse on the display.

[image]

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.98

## To Reproduce

I use the following (from Home Assistant/MQTT) to display a NINA warning:

```yaml
data:
  qos: 2
  retain: false
  payload: |-
    {
      "text": "{{ state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1", "headline") | replace("<br/>", " ") | striptags }}: {{ state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1", "description") | replace("<br/>", " ") | striptags }}",
      "icon": "9335",
      "pushIcon": 2,
      "repeat": 1
    }
  topic: awtrix1/custom/nina
action: mqtt.publish
```

## Expected behavior

Don't know if that's actually possible, but smoother text scrolling for long texts would be superb, especially when a device is used to display warnings.

## Screenshots

(See GIF above)

## Logs

—

## Additional context

—

distanzcheck, 2025-01-09:
Hello, which add-on / integration is this?

distanzcheck, 2025-01-09:
OT: Yes, I also have Nina running with me. Would you make your automations available, display on Ulanzi clock ?  Dirk

Blueforcer, 2025-01-09:
@Moonbase59 how long is your text?
I've tested it with a 380 character string, and it's a very smooth scrolling. Do you use the scroll speed option?

Moonbase59, 2025-01-10:
@Blueforcer Counting above (real) text, I get 340 chars. NINA descriptions can be longer, though. I left scroll speed at default (i.e., never change it). Typical NINA headline+description seem to be around 300–500 chars (after striptags).

Experimenting a little more, I can see the same effect (stutter, hickups) when globally activating the "snow" overlay. Indeed looks a bit like something else hitting in (MQTT app updates?). WiFi reception is quite ok (-55 dBm; about 2 metres from the router), and WiFi total 2.4GHz bandwith usage ~11% (as measured by Fritz!Box router). So nothing unusual here.

Then again, I only use one built-in app (Time), and 7 others in the standard loop, plus NINA and notifications when needed.

`awtrix1/api/loop` currently shows (no NINA warning currently):
```json
{
  "Time": 0,
  "weather": 1,
  "weather_night": 2,
  "weather_fridge": 3,
  "weather_humidity": 4,
  "weather_wind": 5,
  "weather_uv": 6,
  "effect": 7
}
```

My `dev.json` is:
```json
{
  "hostname": "awtrix1",
  "matrix": 0,
  "temp_offset": -10.7,
  "temp_dec_places": 1,
  "hum_offset": 8,
  "min_brightnexx": 1,
  "max_brightness": 100,
  "new_year": true
}
```

My `awtrix1/api/stats`, if it helps:
```json
{
  "bat": 100,
  "bat_raw": 666,
  "type": 0,
  "lux": 0,
  "ldr_raw": 201,
  "ram": 117564,
  "bri": 2,
  "temp": 20.7,
  "hum": 38,
  "uptime": 196768,
  "wifi_signal": -55,
  "messages": 206395,
  "version": "0.98",
  "indicator1": false,
  "indicator2": false,
  "indicator3": false,
  "app": "weather_night",
  "uid": "awtrix_4276d0",
  "matrix": true,
  "ip_address": "192.168.170.102"
}
```

Anything else you want me to check?

---

@distanzcheck Yeah, sure, here goes. You'll have to adapt MQTT topics and entity names, and you need icon \#9335.

NINA Automation (shows/notifies only warning \#1):
```yaml
alias: NINA-Warnung Krumbach
description: ""
triggers:
  - entity_id: binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1
    trigger: state
actions:
  - choose:
      - conditions:
          - condition: state
            entity_id: binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1
            state: "on"
        sequence:
          - data:
              qos: 2
              retain: false
              payload: |-
                {
                  "text": "{{ state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1", "headline") | replace("<br/>", " ") | striptags }}: {{ state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1", "description") | replace("<br/>", " ") | striptags }}",
                  "icon": "9335",
                  "pushIcon": 2,
                  "repeat": 1
                }
              topic: awtrix1/custom/nina
            action: mqtt.publish
          - data:
              qos: 2
              retain: false
              payload: |-
                {
                  "color": [255,0,0]
                }
              topic: awtrix1/indicator1
            action: mqtt.publish
          - action: notify.notify
            metadata: {}
            data:
              data:
                priority: high
              title: >-
                NINA-Warnung ({{
                state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
                "severity") }})
              message: >-
                {{
                state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
                "headline") }}: {{
                state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
                "description") }}

                {{
                state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
                "recommended_actions") }}
      - conditions:
          - condition: state
            entity_id: binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1
            state: "off"
        sequence:
          - data:
              qos: 2
              retain: false
              topic: awtrix1/custom/nina
            action: mqtt.publish
          - data:
              qos: 2
              retain: false
              topic: awtrix1/indicator1
            action: mqtt.publish
          - action: notify.notify
            metadata: {}
            data:
              data:
                priority: high
              title: NINA-Warnung aufgehoben
              message: NINA-Warnung wurde aufgehoben.
```

Code for a simple dashboard Markdown card:
```yaml
type: markdown
content: >
  {{
  as_timestamp(state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
  "sent")) | timestamp_custom('%a %d.%m.%Y %H:%M') }}

  {{ state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
  "sender") }}

  {{ state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
  "severity") }}

  {{
  as_timestamp(state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
  "start")) | timestamp_custom('%a %d.%m.%Y %H:%M') }} – {{
  as_timestamp(state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
  "expires")) | timestamp_custom('%a %d.%m.%Y %H:%M') }}

  ### {{
  state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
  "headline") }}

  **{{
  state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
  "description") }}**

  {{ state_attr("binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1",
  "recommended_actions") }}
visibility:
  - condition: state
    entity: binary_sensor.warning_krumbach_schwaben_st_gunzburg_bayern_1
    state: "on"
title: NINA-Warnung
```

distanzcheck, 2025-01-10:
@Moonbase59 Good morning and thank you for the automations, I will customise these entities and then try them out. Thanks again for that

Moonbase59, 2025-01-10:
@Blueforcer I totally forgot that I also use the [AWTRIX Light Display Card](https://github.com/RDG88/lovelace-awtrix-light-display-card) for remote, proxy-able checking the display, which in turn requests a `awtrix1/sendscreen` every 10 seconds, even when not currently watching the dashboard.

So I tested a little more and disabled that automation, which didn't improve things.

I wonder if we're running into ESP32 performance limits here, or maybe some used library uses blocking code (MQTT?). The only other source of MQTT traffic in my case could be the weather app. Weather data is updated every 15 minutes, but my BLE outdoor & fridge thermometers update more often (every ~15s), resulting in MQTT weather data being updated more often. I'll try to disable the thermometer triggers for a while and see if that improves things.

Would there be any way to further diagnose this, like checking the ESP's CPU load or MQTT traffic?

distanzcheck, 2025-01-11:
@Moonbase59 Is it correct that I have to enter the name of my Roland Sie where it now says abitrix 1? and do I have to add anything else after the name?
the message on the smartphone works like this

Moonbase59, 2025-01-11:
OT: @distanzcheck: I assume "Roland Sie" should have meant "Ulanzi"? For MQTT, instead of my "awtrix1", just use the prefix that is set in your Web UI, on the MQTT tab.

So, as an example, if yours had `awtrix_123456` in there, the nina app topic would be changed from `awtrix1/custom/nina` to `awtrix_123456/custom/nina` and so forth.

Docs here: https://blueforcer.github.io/awtrix3/#/README

distanzcheck, 2025-01-13:
@moonbase59 I might have another cool idea. How can I communicate directly with you so that we don't clutter up this post?

distanzcheck, 2025-01-25:
@Moonbase59 Kann man bei der Automatisierung Warnungen, in meinem Fall "Hochwasser" herausfiltern das diese nicht angezeigt werden. Und wenn eine zweite (stadt_2) da ist das diese auch angezeigt wird.

Blueforcer, 2025-01-25:
Ihr solltet euch woanders treffen fur das Thema. Discord z.b
Hat nichts mit dem issue zu tun

Moonbase59, 2025-01-29:
@Blueforcer Any ideas/answers to my questions above?

Here is a short cellphone video that shows the real display:
https://github.com/user-attachments/assets/c9caec66-56aa-461b-8dd8-b06148366b22

> I wonder if we're running into ESP32 performance limits here, or maybe some used library uses blocking code (MQTT?). The only other source of MQTT traffic in my case could be the weather app. Weather data is updated every 15 minutes, but my BLE outdoor & fridge thermometers update more often (every ~15s), resulting in MQTT weather data being updated more often. I'll try to disable the thermometer triggers for a while and see if that improves things.
> 
> Would there be any way to further diagnose this, like checking the ESP's CPU load or MQTT traffic?
> 
> **EDIT:** Reducing the weather (temperature) data MQTT traffic from every 15s to every 15min reduced the scroll stutter a lot, but not completely. In my case, this can't be a permanent solution, though. So the question remains: Can this be somehow improved/decoupled?

@distanzcheck Sorry, I can't currently work on anything new. The AWTRIX and Home Assistant documentation should answer all questions. The [Home Assistant Community Forum](https://community.home-assistant.io/) might be another good place to ask questions, lots of helpful people there!

Blueforcer, 2025-03-03:
> In my case, this can't be a permanent solution, though. So the question remains: Can this be somehow improved/decoupled?

No, because it cannot be reproduced and you seem to be the only one among thousands of users who has this problem.
either your Ulanzi  is defective in some way, or you are doing something with the APIs that nobody else is doing.
I am firing my Ulanzi at 80 messages per minute without any performance problems

Moonbase59, 2025-03-07:
Hard evidence against my setup, it seems ;-)

Have to investigate further then. I somehow suspect my "weather app", which updates quite often (on any status or attribute change in HA) and sends icon data in the payload. Maybe I can streamline that a little and pre-load icons onto the Awtrix.