Blueforcer/awtrix3 #676: [BUG] after update to 0.98 TC001 not accessible anymore

UweW, 2025-01-13:
# Bug report

## Describe the bug
after update to 0.98 TC001 not accessible anymore.
Accesspoint still see TC001 connected with the last IP provided by dhcp server.
Multiple rebbots did not help

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.98

## To Reproduce

Steps to reproduce the behavior:
- download firmware
- update firmware via webinterface
- after reboot, current IP is shown
- not reachable under this IP
- not connection to MQTT server

## Expected behavior
accessable via network

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

nmap does not show any open ports or any ping response

etienne72230, 2025-01-15:
Same issue. Clock seems to work very slow (clock dot update every 10sec) with a yellow pixel at bottom left corner. Before update clock work perfectly with wifi and mqtt.

Blueforcer, 2025-01-15:
That's not the same issue. In your case Awtrix cannot connect to your MQTT broker.

For the topic issue im currently searching for an possible error, but since nothing has changed according to any kind of connection method and only one issue of that kind out of thousands users, it's currently hard to tell.

codedmind, 2025-02-22:
@Blueforcer i'm having. the same issue.
I change the firmware using ulanzi webpage, because using flasher is always failing (tried in a mac and windows pc, multiple cables)

Now i cannot do anything.. if i try connect again via webpage to see the logs the ulanzi beep 3 times a never connect, is always in ap mode, when i try to change the ssid it's too slow and returns a lot of errors in the webpage, can't do anything.

Any suggestion how to solve this? Can i help send you some logs ?

coulson84, 2025-02-28:
I am seeing this behaviour on both clocks I have. I updated one today and can no longer access the webserver at the clock's IP address. I had another TC001 that was still in the box so got that one out and flashed it to see if it was just something wrong with the other clock but the web UI is not accessible for either of them.  Here are the logs from one of the clocks when it is reset

```
Connecting to coulpops
*.**.**.*
Connected to Wifi! IP address: 192.168.178.50
[2043] [startLittleFS]: Starting filesystem
[2063] [startLittleFS]: Filesystem started
[2063] [loadSettings]: Loading Usersettings
[2065] [getID]: Starting filesystem
[2065] [loadDevSettings]: Loading Devsettings
[2077] [loadDevSettings]: Devsettings not found
[2077] [setup]: Setup periphery
[2092] [setup]: SHT31 sensor detected
[2094] [loadSettings]: Webserver configuration file not exist
[2095] [setMatrixLayout]: Set matrix layout to 0
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to coulpops
.[3051] [setup]: My IP: 192.168.178.50
[5398] [setup]: Webserver loaded
[18678] [setCurrentApp]: Publish current app Temperature
[26090] [setCurrentApp]: Publish current app Humidity
```

markf, 2025-03-01:
I'm having the same problem. Same output from the logs as well - IP address is the one my DHCP server is set to assign it, and if I change that, the clock will receive the new IP address. However, cannot ping the assigned address nor can I telnet to port 80.

UrsKobler, 2025-04-17:
After upgrading the firmware from Version 0.96 to 0.98, I noticed that synchronization between the Wi-Fi router and the display (Ulanzi Awtrix Smart Pixel Clock 2882) is temporarily unavailable.

**_IMPORTANT_**:
**_1. Usecase-A (permanent Wi-Fi operation)_**
If the Wi-Fi router is turned on before turning on the display, there is **NO connection issue**.

**_2. Usecase-B (Wi-Fi operation with schedule)_**
If the Wi-Fi router is turned off and then back on after successfully communicating with the display, the display does not always synchronize, even if the distance to the Wi-Fi router is 2 meters and the signal strength is excellent!

I have been using the display with the Awtrix-Light firmware in the iobroker environment for two years. The display has a static IP address of 10.0.0.19.

For energy efficiency reasons, all unused devices in the smart home are automatically turned off and turned back on when residents are present. The same applies to all Wi-Fi routers and access points, which are turned off at night and turned back on during the day.
To ensure that the display's internal battery is cyclically discharged and charged, a controller monitors the battery charge level and switches the USB power supply at the switching thresholds of 85% and 40%.

@Blueforcer
Would it be possible to use a protocol sniffer to verify the cause of this synchronization issue? All you would have to do is turn the Wi-Fi router on and off. 

Thank you for the excellent firmware.

UrsKobler, 2025-05-05:
In the meantime, I've investigated the symptom identified by @UweW more closely. I've noticed the following phenomenon:

Whenever the display freezes, the Wi-Fi access point/router connected to the display occasionally crashes due to a suspected protocol inconsistency, so that simply restarting the Ulanzi-TC001 display is no longer sufficient to establish a successful connection. In such a case, the Wi-Fi access point/router must also be restarted.

However, I don't believe the original cause of the described problem is communication (HTTP request API), but rather a suspected memory leak in the program code responsible for processing the animated GIF images.

To reproduce the freezing of the display, I use the following GIF files, which are activated or deactivated depending on the application context (reaching a specific outside temperature, reaching a specific air pressure, reaching a specific PV power, etc.).

[image]

I use the Ulanzi TC001 display in the iobroker environment, where the Awtrix Light Adapter v1.7.0 (from the developer @haus-automatisierung) is used. This adapter is based on the HTTP request API interface.

**_My SmartHome system requirements_**:

Platform Operating system: **Linux**
Architecture: **Arm64**
CPUs: **4**
Speed: **2400 MHz**
Model: C**ortex-A76**
RAM: **7.8 GB**
Node.js: **v20.19.1**
NPM: **10.8.2**

I would appreciate it if @Blueforcer could investigate my suspicion regarding the error in more detail.

Thanks very much
Kobi

Blueforcer, 2025-09-24:
@UrsKobler 
im not able to use any third party plugin but i finally tried all youre icons without any issues.
Do you log the free_ram from the /stats to see if there is a memory leak?
With a constant memory leak the whole ESP should crash at some time.

xstrex, 2026-01-12:
Since this is clearly a bug in 0.98 how do I install 0.97?