Blueforcer/awtrix3 #677: [BUG] I don't know what to make of these bat_raw values

unn, 2025-01-14:
# Bug report

## Describe the bug
I was seeing some weirdness with the clocks battery reporting. After reviewing the issue queues, I found your comments on https://github.com/Blueforcer/awtrix3/issues/624#issuecomment-2444218305 and then https://github.com/Blueforcer/awtrix3/issues/568 so I set off to monitor the reported `bat_raw` levels in HA.

I decided that I'd fully charge the battery to grab the `max_battery` then fully drain for `min_battery`.  This informed the values for my `dev.json` file of `"min_battery": 787, "max_battery": 889,`. These seemed a little too close, but after a few minutes of run time, the battery levels seemed to be reporting more sanely. However, fast forward a few hours, the battery is on the charger and now reporting 0% charged, with a `bat_raw` value in the 1xx range. I removed the entirety of the `dev.json` file and rebooted the device.

I left the device on the charger until the light on the rear turned green, unplugged it and went to bed. Keep in mind, it's on it's internal battery and unplugged. Here is what I'm looking at this morning. [image]

I know you've been pretty down on Ulanzi's battery quality, but... this just seems odd that it would report an increase in `bat_raw` while not being on a charge and then see no decrease in `bat_raw` for a number of hours.

I guess it's possible I could have a really defective battery, but what do you think? I'm considering flashing the OEM firmware to see if it exhibits the same weirdness and returning. But maybe there is some other stone to look under?

Thanks in advance.

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.98

luebbe, 2025-01-15:
There ist also some interesting information about battery monitioring in the following thread: #391.
But I have to admit that your batt_raw curve looks very weird. I never had values as low as you. A sensible range for my battery is min_battery: 600 and max_battery unchanged. Running awtrix 0.96.
My Ulanzi is now a bit more than one year old, so you can consider the possibility that they changed the hardware in the mean time.

unn, 2025-01-15:
Yeah, that is where my head went as well. Unfortunately, it doesn't seem like there is an obvious hardware revision on the exterior of the unit.

unn, 2025-01-22:
I returned the clock and got a new one. The new one seems to have a battery graph that doesn't increase bat_raw when not on a charge.

closed 2025-01-22 by unn