Blueforcer/awtrix3 #682: [FEATURE REQUEST]

thuibregtse, 2025-01-20:
# Feature Request

### Is your feature request related to a problem?

3 available "matrix" values in dev.json do not seem to support the DIY WESIRI 8x32 pixel layout.   After flash and reboot, I always get random pixels in the center 2/3 of the display

### Describe the solution / feature you'd like

Support for additional pixel layouts if needed.  Per one of the online reviews, "you will find that all LEDs are connected in series from left upper corner to the right bottom corner"

### Describe alternatives you've considered

Before AWTRIX I had this display working with an 8266 and WLED, including text.
Per the awtrix DIY docs, I created a dev.json file in the top of the src directory in vscode.
I tried all available values for "matrix" in the json file, rebuilt, flashed successfully, and rebooted each time.   

I also looked at the source code....experimented with MATRIX_LAYOUT values 

[image]

in case the dev.json was not doing what it needed to.

### Additional context

Per the docs, display is being driven by GPIO32 of a ESP32 dev board.  I have a solid 5A power supply driving the ESP32 and the display.

I would be glad to try other suggested troubleshooting ideas.

casatift, 2025-01-20:
encountered the same issue. The panels supported by Awtrix are connected end-to-head in a vertical arrangement, but each column of this panel is arranged from top to bottom."

thuibregtse, 2025-01-20:
@casatift, thank you for confirming.

I forgot to mention that I did get this display working with @lubeda's Ehmtx, so maybe there are some code ideas there.

[image]

Blueforcer, 2025-01-20:
if you can tell me the correct alignment i can add it. if youre fimilar with platformIO programming you can play with it:
https://github.com/Blueforcer/awtrix3/blob/f12513abce8e2b9fb8754df88258e0f03cad8fdd/src/DisplayManager.cpp#L1674

thuibregtse, 2025-01-20:
Here's what I found in another online review of the display (vendor docs are sparse):

"The individual LED strips in this panel were upside down, which results in a top row R-to-L serpentine pattern instead of the correct top row L-to-R serpentine pattern. This results in a mirrored image..."

As soon as I finish shoveling out from the recent snowstorm, I will put an 8266 and WLED on.  I'll iterate through # of LEDs, etc. to answer this the brute-force way.    

Thanks for all the great work on awtrix!  I'm looking forward to getting this going with HA.

thuibregtse, 2025-01-20:
Here is the command I'm using to upload:
esptool --port COM9 write_flash 0x00000 firmware.bin

Firmware uploads successfully.   I unplugged and then powered up the unit.  Result is a dark screen and no Wifi AP.   Hardware is still good if I re-run with the web flasher (random pixels displayed)

 Is my command incorrect, or should I go ahead with my WLED experiment?

Blueforcer, 2025-01-20:
0x00000 is incorrect as offset. the firmware startes at 0x65536
Easiest way is to just upload the firmware with the webinterface

thuibregtse, 2025-01-20:
I took a few minutes to check the pixel layout.  It looks like it's serpentine in the short direction

[image]

[image]

I haven't been able to upload the firmware with the offset from the command line.  If I use the web flasher, where would it locate the new firmware on a Windows 10 system?  Is it assuming a location?

Blueforcer, 2025-01-20:
Webflasher works with the path to the github binary, you cannot upload a custom firmware.
Open the awtrix webinterface (with the released firmware), klick on update and upload the firmware bin

thuibregtse, 2025-01-20:
I installed released firmware, connected to my Wifi, and then used the UI to update to the new firmware.  Rebooted.
Still random pixels scrolling around.  New firmware is definitely there...pixels now show up on all parts of the matrix.

Blueforcer, 2025-01-20:
Did you set the matrix to layout 3?

thuibregtse, 2025-01-20:
I’ll be away for a couple of hours.  Do I put the dev.json in the same
folder as the binary?  I’ll try as soon as I get back home.

Thank you

Blueforcer, 2025-01-20:
wait a moment... 
Now I have understood yout first post...
You have to create the dev.json in the filemanager of AWTRIX, VScode has nothing to do with it.
Especially not if you work with the webflasher.

thuibregtse, 2025-01-20:
Ah, <that> file manager.  Blinding Flash of the Obvious (BFO)

Still not there yet.   Firmware has been updated to Awtrix 3 0.98
Root of awtrix file system has a dev.json as shown  ("matrix":3)

[image]

Everything has been rebooted.
Here's a live view vs what's on the matrix:

[image]

It looks like rows and columns may be heading in different directions

Blueforcer, 2025-01-20:
And the file manager show you the error with the red dot. The file is not valid

thuibregtse, 2025-01-20:
Embarrassing to publicise one's typos....fixed.

[image]

After a save and reboot, the results are different (see vs above)

[image]

Blueforcer, 2025-01-20:
Now test 0 1 and 2

thuibregtse, 2025-01-20:
Dang.   "2" worked.   Sorry about the wild goose chase!

Blueforcer, 2025-01-20:
No problem!
Have fun with Awtrix

closed 2025-01-20 by Blueforcer