Blueforcer/awtrix3 #686: Awtrix  3 - DF Player no sound output

pwasserf, 2025-01-24:
Hello,

I have updated my Awtrix 2 to the ESP 32 mini and software version 0.98.

So far so good, however I am try to connect the DF Player board to the ESP as shown in the Hardware Section on gitub -> Getting Started.

Unfortunately no MP3 file can be triggert, no sound output.

Can you please be so kind and confrim the TX and RX pins on the ESP32 mini and to be on the save side an expample on the syntax to to trigger the the first file MP3 file on the SD.

I am using: msg.payload = {"sound":"0001"} and send a HTTP POST:http://192.168.100.94/api/sound.

Thanks for your support.

Peter

Blueforcer, 2025-01-24:
Did you enabled the Dfplayer via dev.json?

Blueforcer, 2025-01-24:
https://blueforcer.github.io/awtrix3/#/dev

pwasserf, 2025-01-24:
Thanks a lot, I have overseen this now it working :-)

closed 2025-01-24 by pwasserf