Blueforcer/awtrix3 #693: Login to AP Mode

PeterSpie, 2025-02-11:
Hello,
I have just flashed my Ulanzi with firmware 0.98.
When I want to connect to the AP, to set the Wifi, it aks me to login. I know the password is 12345678, but there is no where to enter this password.

[image]

The network says "Don' use this network", or "Use this network in its current state".

[image]

What am I missing/doing wrong?

Regards, Peter

Blueforcer, 2025-02-11:
According to the screenshots you're already connected with Awtrix.
12345678 is the wifi password for the Awtrix hotspot.

PeterSpie, 2025-02-11:
Well, It just doesn't work. I get errors stating the credentials are incorrect, but I am 100% positive they are correct.
One time, it said "connected" on my mobile phone, but then after a few seconds the display turned to "AP Mode" again.

I am not a developer, and in the log I seem some errors. But maybe they are not related to this problem.

`[2041] [startLittleFS]: Starting filesystem`
`[2061] [startLittleFS]: Filesystem started`
`[2061] [loadSettings]: Loading Usersettings`
`[2063] [getID]: Starting filesystem`
`[2063] [loadDevSettings]: Loading Devsettings`
`[2075] [loadDevSettings]: Devsettings not found`
`[2075] [setup]: Setup periphery`
`[2090] [setup]: SHT31 sensor detected`
`[2092] [loadSettings]: Webserver configuration file not exist`
`[2093] [setMatrixLayout]: Set matrix layout to 0`
`Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256`
`AP mode.`
`Server IP address: 192.168.4.1`
`[2757] [setup]: My IP: 192.168.4.1`

Do you know what could be wrong?

Blueforcer, 2025-02-11:
There is no error in the log.
Awtrix doesn't try to connect, it seems that you haven't entered any wifi credentials.
Enter your home WiFi SSID and Password, then click connect to ssid. After that you will get a pop-up with the new IP address.
Make sure you're connection to a 2,4ghz wifi. 5ghz is not supported