Blueforcer/awtrix3 #698: ESP32 awtrix 0.96

valbuz, 2025-02-15:
I have a orig. Ulanzi with awtrix and a own flashed esp32 with awtrix 0.96.

with my own esp32 i can't download any icon. the prev is ok, but download is not working.

i'm flash the new fw with erase before.

what's going wrong?
thanks for any help.

pat

Blueforcer, 2025-02-15:
The current firmware version is 0.98.
Issues for old versions doesn't make sense.
Please try to update first.

valbuz, 2025-02-15:
Strange...
my first install on the esp was with 0.98 and it doens't work.
after that i downgrade to 0.96 ( like my orig ulanzi) and it doesn't also work.
so i wrote this issue..  
now i flash again with 0.98 and it works!
not my day...
thanks a lot

closed 2025-02-15 by valbuz