Blueforcer/awtrix3 #704: [BUG] HA entities not available

mrcl248, 2025-02-26:
I recently flashed two TC001 with the newest version.
One is working pretty fine, but the other one shows no entities in Home Assistant, MQTT is working because the logs are working and display every app change.

I repeated the flashing several times and erased the device in HA as well, the auto discovery works but the entities are still unavailable.

What can cause the problem and what can I do?

Blueforcer, 2025-02-26:
If one is working and the other not with the same version it's rarely a software bug.
Remove the device from your HA mqtt device list and restart awtrix

mrcl248, 2025-02-26:
> If one is working and the other not with the same version it's rarely a software bug. Try to remove the device from your HA mqtt device list and restart awtrix

Thanks for the fast reply.
I removed it and restarted, it's the same. I found out, that my new prefix, I entered in awtrix isn't published to HA, it's listed with the original device name.

Blueforcer, 2025-02-26:
Did you flashed the Ulanzi with erase before?
If you change anything in the Webinterface, you need to restart awtrix after saving, for the time beeing

mrcl248, 2025-02-26:
Yes, of course.
If I change the name to another, save and restart, the old name is still shown in HA.