Blueforcer/awtrix3 #706: [BUG] Temperature sensor wont report to HA Via MQTT

MrPresident2, 2025-03-03:
I have a weird problem.
I added a temp sensor to my Awtrix build, the devices show on screen the temperature and humidity, but while seeding it to HA via MQTT I see information from the first time the MQTT was connected. So, one update a few days ago. everything else (MQTT) works, I can see the status of Awtrix, the app running, and all the other information.

Awtrix Version: 0.98 (latest).
Custom build - ESP32 and LED Matrix

dev.json
```{
    "matrix":2,
    "temp_offset":0,
    "stats_interval": 10000
} ```

Pics:

![Image](https://github.com/user-attachments/assets/308c1690-128f-41b6-a5c9-cb59d3e91842)

![Image](https://github.com/user-attachments/assets/5fced4e7-c346-42dd-ac14-cabfdb479605)

xylorin, 2025-04-06:
got the same issue, e.g. adding a temperature sensor reprots only the temp. once and it stays the same

dlizcano, 2025-04-07:
I have the same issue

MrPresident2, 2025-04-09:
Still no answer about it.