Blueforcer/awtrix3 #710: [BUG] MQTT: when setting indicator 2 or 3, Awtrix doesn't update "name/id/XXXXXX_ind2/stat_t" on MQTT

neurolit, 2025-03-12:
# Bug report

## Describe the bug

When setting indicator 2 or 3 by publishing a MQTT message on topic `awtrix_XXXXXX/indicator2` (or `awtrix_XXXXXX/indicator3`), Awtrix doesn't publish on topic `awtrix_XXXXXX/id/XXXXXX_ind2/stat_t` (or `awtrix_XXXXXX/id/XXXXXX_ind3/stat_t`. However, it works for indicator 1.

``` sh
mosquitto_pub -t 'awtrix_7ac880/indicator1' -m '{ "color": [53, 132, 228], "blink": 0}'
mosquitto_pub -t 'awtrix_7ac880/indicator2' -m '{ "color": [53, 132, 228], "blink": 0}'
mosquitto_pub -t 'awtrix_7ac880/indicator3' -m '{ "color": [53, 132, 228], "blink": 0}'
mosquitto_pub -t 'awtrix_7ac880/indicator1' -m '{ "color": "0" }'
mosquitto_pub -t 'awtrix_7ac880/indicator2' -m '{ "color": "0" }'
mosquitto_pub -t 'awtrix_7ac880/indicator3' -m '{ "color": "0" }'
```

Here is what is published:

``` sh
mosquitto_sub -t 'awtrix_7ac880/indicator1' -t 'awtrix_7ac880/4c75257ac880/7ac880_ind1/stat_t' -t 'awtrix_7ac880/indicator2' -t 'awtrix_7ac880/4c75257ac880/7ac880_ind2/stat_t' -t 'awtrix_7ac880/indicator3' -t 'awtrix_7ac880/4c75257ac880/7ac880_ind3/stat_t' -v

# works for indicator 1
awtrix_7ac880/indicator1 { "color": [53, 132, 228], "blink": 0}
awtrix_7ac880/4c75257ac880/7ac880_ind1/stat_t ON
# no message for indicator 2
awtrix_7ac880/indicator2 { "color": [53, 132, 228], "blink": 0}
# no message for indicator 3
awtrix_7ac880/indicator3 { "color": [53, 132, 228], "blink": 0}
# works for indicator 1
awtrix_7ac880/indicator1 { "color": "0" }
awtrix_7ac880/4c75257ac880/7ac880_ind1/stat_t OFF
# no message for indicator 2
awtrix_7ac880/indicator2 { "color": "0" }
# no message for indicator 3
awtrix_7ac880/indicator3 { "color": "0" }
```

Please note that all the indicators appear or disappear on the Ulanzi screen as expected.

What could I do?

Blueforcer, 2025-03-12:
Are you using HomeAssistant or why you trying to read manually from this topic?

Blueforcer, 2025-04-07:
Sorry for the late response.

Currently im unable to reproduce it

[image]

Switches in HA turn on and off according to the indicator message. Works in all 3.

Please try to turn off AWTRIX, remove the device from homeassistant and start AWTRIX again.