Blueforcer/awtrix3 #711: [BUG] TC001 ignore dev.json

mluckau, 2025-03-15:
# Bug report

## Describe the bug

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0,98

## To Reproduce

Steps to reproduce the behavior:

Create a dev.json with the file manager, enter the settings. dev.json is ignored.

## Expected behavior

The display is rotated so that the ulanzi can be hung upside down.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Hello, I have flashed a TC001 with awtrix 0.98, because I hung it upside down under a shelf I wanted to rotate the display. I created a “dev.json” with the file manager with the following content:

````
 {
  "hostname": "awtrix_xxxxxx",
  "rotate_screen":true,
  "bootsound":true
}
````

However, the settings are not applied and I can no longer delete the file and it can no longer be changed. No matter what I write in it, after I have saved the file and I reload the page, it is back to the old state, deleting the file is also no longer possible.

Blueforcer, 2025-03-15:
Your posted Json and the web editor gives you the hint to your fault.. remove the leading space.

Blueforcer, 2025-03-15:
Well 1 minute ago there was a different issue description 😳

mluckau, 2025-03-15:
> Your posted Json and the web editor gives you the hint to your fault.. remove the leading space.

That's part of the problem, I can neither edit nor delete the file.

Blueforcer, 2025-03-15:
Is your flash space full? That's the only reason I could imagine. Otherwise maybe the file is completely corrupted so you can try to reflash Awtrix. But that never happens before

mluckau, 2025-03-18:
Hello,

the problem really seems to have been a memory problem. I tried to reflash it but that didn't help. I then waited until the battery was completely empty and the esp was completely without voltage. Then I flashed awtrix again and everything worked as usual. I may remove the battery so that a reset is possible at any time, I only operate it with a permanent power supply anyway.

closed 2025-03-18 by mluckau