Blueforcer/awtrix3 #712: La Metrix icons are not working

lucabacchipg, 2025-03-16:
Hi, has anything changed on La Metric?
The icons are no longer downloadable, and if downloaded directly, they freeze the display.
Has something changed?
Thanks.

puch1951, 2025-04-08:
I had the same problem.
Would be nice if the download will work again.

Blueforcer, 2025-04-08:
If tested several icons. Each is downloaded and working correctly.
Make sure you're running the latest firmware

puch1951, 2025-04-11:
Hallo Blueforcer,

thanks for the fast answer. I have installed AWTRIX 3  0.96 (great project!).

The  preview with the Icon ID works fine. When i click on „Download“ nothing happens at all.

 

Regards

Udo

Blueforcer, 2025-04-11:
Latest version is 0.98 from January.
This version fixes the problem.
Please make sure to use the latest firmware before open an issue.