Blueforcer/awtrix3 #716: [BUG]

vins84, 2025-03-23:
# Bug report

When battery dies Im unable to set via UI the correct date and time for the UK after turning it back on. It seems like it goes to default 01/01 00:00

## Describe the bug

Once battery runs out of juice date and time seems to get wiped/set to default value. Setting time zone to UK via UI does nothing. Setting other timezones seem to work (at least for the time part) but as soon as I change it back to UK i.e. GMT0BST,M3.5.0/1,M10.5.0 time and date isnt getting applied. Ive also tried UTC0 and GMT0 with no luck.

if the clock is up and running restarting it doesnt wipe the time settings. It only seem to happen when battery runs out. 

The only way to fix it is to reflash device. 

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [ eg. v0.98 ]

## To Reproduce

Flash and setup device time zone to GMT0BST,M3.5.0/1,M10.5.0. Let the battery run out. Turn the clock back on by holding two arrows, notice datetime to be 1/01 00:00.

## Expected behavior

Timezone to be applied on every restart including when battery dies. 

## Screenshots

## Logs

## Additional context

Is it possible to manually trigger calling NTP server again to set the time again or does it only happen during initial flash?

Blueforcer, 2025-03-23:
NTP Sync doenst happen during flash, its done on every boot.
AWTRIX uses the the official esp32-hal-time libary
https://github.com/Blueforcer/awtrix3/blob/f12513abce8e2b9fb8754df88258e0f03cad8fdd/src/ServerManager.cpp#L275

There is no difference between a dead battery or switching off directly using the buttons. When you switch the ulanzi back on, is your time zone still locked or reset to factory settings?
Have you flashed awtrix via the WEbflasher and also selected “erase” beforehand?
I am not aware of any similar case so far.

vins84, 2025-03-23:
Hmm weird. Maybe it is having issue with this specific timezone then. The settings are still correct when I switch it back on. And yes I always flash using WebFlasher with "erase" option selected to eliminate any potential issues.

Blueforcer, 2025-03-23:
Is Awtrix accessible after restart?
Maybe the NTP client is unable to connect to the NTP server

vins84, 2025-03-23:
Yes everything works as expected, at least I have not noticed any other issues.

"Maybe the NTP client is unable to connect to the NTP server" could be but it is weird that it works immediately after reflashing it.

Blueforcer, 2025-03-23:
Is this also the case when you setup your Timezone an turn the clock Off and on with the two buttons?

vins84, 2025-03-23:
I did mention it in the description of the bug but I shouldve been clearer...no it doesnt happen when I manually turn if off and back on using buttons. It only happens when battery dies. Wifi in my kitchen is quite weak but I have it on my desk right now, battery should die soon so I will check again but Im pretty sure it happened to me when I initially played with it before moving it to the kitchen

vins84, 2025-03-23:
so it died and yet again clock shows 01/01 00:00. I dont know if it is related but Ive noticed that during last few minutes it kept restarting constantly (literally 30 times) presumably due to the low battery.

m4ding, 2025-03-26:
Maybe my issue is connected ti this: my ulanzi is showing the wrong time (offset of 1h:10m). I tried rebooting, downgrade to 0.97, verifying the time zone, setting another ntp. Doesnt help.

vins84, 2025-03-26:
the only way I found to correct it is to flash it again (I always tick "erase" device so cant comment if it makes any difference). It would be good to have a MQTT endpoint for refreshing dateTime

Blueforcer, 2025-04-07:
Im unable to reproduce it.

ive tested it with "AEST-10AEDT,M10.1.0,M4.1.0/3" wich is 01:53 now
(GMT0BST,M3.5.0/1,M10.5.0) doesnt exist according to https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

then i discharged the battery, and after turning it on again, the correct time is back.

I recently had someone in my awtrix discord who had a similar problem. In his case, the PiHole was responsible for the clock not being able to connect to the internet.

vins84, 2025-04-09:
GMT0BST,M3.5.0/1,M10.5.0 Does exist on https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

[image]

Blueforcer, 2025-04-16:
Im still unable to reproduce it. Also, thousands of other users don't seem to have this problem. 
As I said in my last post, you need to check whether something is blocking AWTRIX's internet connection.

bjakubski, 2025-04-22:
I left recently my TC001 powered off and disconnected for couple of days and after powering on (left+right) it was unable to get time and date via ntp. (as far as I know battery was still charged).
Just like described in this issue here: no timezone changes, no NTP server changes nor reboots were able to fix this. I did not try reflashing.
But I was able to work around the issue by using *IP address* as "NTP server" - save&reboot and it works!
Nothing really changed in my network so I'm not sure why it stopped working.
@snake302 Maybe this will help you?

eku, 2025-04-22:
> But I was able to work around the issue by using _IP address_ as "NTP server" - save&reboot and it works!

This may indicate a sequencing problem with the initialization of the services. NTP before DNS?

bjakubski, 2025-04-22:
> Hey [@bjakubski](https://github.com/bjakubski), thank you for the solution. Do you mean that you entered your TC001 device's IP address, or which IP address did you use as the NTP server?
> 

Just see what IPs `de.pool.ntp.org` resolves to and take one of them. For example maybe try `178.63.9.212`

eku, 2025-04-22:
As far as I understand the source Arduino code

https://github.com/espressif/arduino-esp32/blob/99520f66f61d3c88014356ee96e1db89b1931ee4/cores/esp32/esp32-hal-time.c#L67-L78

the NTP server is contacted on the background.

The result is read only once in ServerManager_::setup()

https://github.com/Blueforcer/awtrix3/blob/2f65f1f456dd1e0e916b948b6753ef2ec99dda5f/src/ServerManager.cpp#L273-L275

There might be a race condition on slow DNS resolution.

udo1toni, 2025-04-23:
Maybe DNS forwarding is the problem? So, try to restart your router.

To be honest, DNS for time server never worked for me (router is OPNSense, DNS is no issue in my LAN, may it be local or global), so I have the router IP set in AWTrix since day one :)

But on the other hand, I always set the mqtt broker with the FQDN and never had an issue with that...

NickSutton, 2025-05-26:
I'm also suffering this problem. The time shown on the display is an hour behind since the clock change and no amount of fiddling with timezone values or NTP addresses seems to be able to fix it!

vins84, 2025-07-03:
It looks like the issue for me was a static IP. Ive always set the static IP to be the same one as the one it connects to wifi with. Good few weeks ago the clock failed to set the time again after a restart so when flashing it again I decided not to set the static and it has bee fine since then. The battery was drained twice since then (due to a temperamental zigbee usb switch not triggering correctly, nothing to do with Awtrix) and yet after turning it back on, in the same location as before; the time was set absolutely fine.