Blueforcer/awtrix3 #718: [BUG] Temp_Offset is not working properly

AndreasvonSchleinitz, 2025-04-06:
# Bug report

Hi, I have the problem with the Temp_Offset function, no matter what negative number I enter it is ignored. All other functions (e.g. Hum_Offset) work without problems. The whole function is:
{
  ‘temp_dec_places":1,
  ‘temp_offset":-9,
  ‘hum_offset":37
}
and is located in a dev.json file. But it does not work. I have noticed that a positive offset is executed, but a negative one is not.

I would be glad if one of you knows some advice.

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - v 0.98

## To Reproduce

Steps to reproduce the behavior:Just saving a dev.json 

## Expected behavior

Showing the offset on the device

Blueforcer, 2025-04-07:
The json is invalid
youre using ‘ instead of "

`{
  "temp_dec_places": 1,
  "temp_offset": -30,
  "hum_offset": 0
}`

works as it should

AndreasvonSchleinitz, 2025-04-07:
Hello Stephan,

surprisingly, it worked once when I copied your code line directly and did not change the values. When I then adjusted the temperature offset to -7, this change was ignored. Hum and decimal places, however, were adjusted.

All the best,
Andi

Blueforcer, 2025-04-16:
> I'm having the same problem. Temperature was on 29 degrees, seems I have to use -18 to get it to 20 degrees...

i dont think its the same problem, because ýour changes has effects.

you need to note, that by default the offset is -9
So when it showing 20° its actually 29°. If you set the offset to -9, its the same as the default value.
so -9 is your zeropoint in this case and you need to calculate from this. OR set it to 0 and then calcualte your offset.

But you must also understand that you will NEVER get the right temperature. The sensor sits in the housing and is heated to different degrees depending on the number of pixels, brightness, etc.