Blueforcer/awtrix3 #719: [BUG] Time incorrectly shown, time zone change has no effect + display issues

derechtegraf1, 2025-04-06:
# Bug report

## Describe the bug

I have flashed AWTRIX on a Ulanzi TC001. The clock is "fed" via NodeRed; http push requests are shown correctly and without any issues.
After setting up the flows, I wanted to send the clock to deep sleep during the night; while sending it to sleep via flow worked, I couldn't get it to wake up again.

So instead of the deep sleep, I decided to simply turn display brightness to 0 during the night and back to 50% in the morning.
This works via flow, but when I checked on the clock the next morning, the time display was off. It seemed as if the time was set to 00:00, 01.01., Monday when it turned back on.

I have since tried several restarts, changing the time server, changing the time zone, and flashing again via "update firmware".
The time display is still broken.

In addition, the clock shows random graphic issues: Flickering, random pixels showing wrong colors, etc. This happens without a specifc pattern, some screens show the errors, some don't; some minutes later the screens that previously had no errors, now have some. There are also periods without any errors at all.

## Additional information

- Devices involved:
  - Model: Ulanzi TC001
  - awtrix3 version: v0.98

## To Reproduce

Steps to reproduce the behavior:
Unclear; error is continuously present.

## Expected behavior

Correct display of time in my selected time zone. No graphical issues on screen.

## Screenshots

[image]
[image]

Blueforcer, 2025-04-07:
That's an hardware issue.
That random pixels are not caused by the software.
Since the time problem seems to be unique to you, I assume that the device has a general problem.
I recommend getting a replacement first

derechtegraf1, 2025-04-07:
Thanks for the feedback. I have ordered a replacement and will test the same flow with the new device.
The coincidence is strange though, as the device worked flawlessly after setup and before adding the sleep / brightness adjustion flow.