Blueforcer/awtrix3 #721: [FEATURE REQUEST] Introduce ATRANS mode that alternates between clock and custom apps

Hypfer, 2025-04-10:
# Feature Request

### Is your feature request related to a problem?

I'm using a bunch of flashed ulanzis as clocks that display quite a lot of other things.
This can be quite annoying, because at times, you need to wait for multiple minutes until the clock is being displayed again.

This makes the smartclock bad at being a clock, which is not ideal.

### Describe the solution / feature you'd like

I'd like there to be a mode where the automatic transition works like this:

`Clock -> App1 -> Clock -> App2 -> Clock -> App3 -> Clock -> App4 -> Clock -> ...`

### Describe alternatives you've considered

I suppose this could be done via external orchestration, however it honestly feels like the firmware should be doing it.

### Additional context

N/A

Blueforcer, 2025-04-10:
The request is a bit more specialized, and will be very difficult to implement in such a generic way that it can be used by everyone. Because not everyone wants to have the time app, for example. That's why you have all required API commands to build your own loop orchestra.

But..

You can try to use an unofficial HTTP API.
Send a Json array of the apps to /api/reorder

["app1","Time","app2","Time",...]

This API is highly experimental and therefore not documented.

- make sure to use the correct name
- you will need to send it after each restart, after all apps exists, and if you're adding or removing apps.

Hypfer, 2025-04-10:
> The request is a bit more specialized, and will be very difficult to implement in such a generic way that it can be used by everyone. 

True, but does it have to be?
I think it would be reasonable to say that the thing is a smartclock and thus the clock part receives special treatment.
Also, after all, those inbuilt apps are special already by itself.

I don't think that it would be inconsistent to do so nor would I worry about someone being unhappy that it's not generic enough.

It's definitely an admirable goal to be _this_ generic, but it can also be rather paralyzing, because while trying to solve fictional problems, solutions to real ones get postponed indefinitely or are simply not really possible.

The question of if you would _want_ to do that is of course something else entirely.

My advice would just be to not worry about opinions of random irrelevant people that have incoherent ideas; _if_ that is something you might be worrying about.

Not saying that because it would serve me here but more as a general observation.

If it is your own hearts desire to be this generic then please ignore what I said :)
I just know how.. uh.. intimidating the maintainer role can be.

> You can try to use an unofficial HTTP API.

Thanks, though I think that doesn't change much over doing _all_ the orchestration externally, because that still means regularly polling all apps, calculating the expected order and then pushing the order again and doing that for all clocks all the time.

At that point, you might as well just have the full scheduler externally.

Hypfer, 2025-04-11:
Okay, after having given it some more thought, I came up with my own use-case that would require this to be more generic. :-)

Specifically, a kitchen timer. That thing should also have the same priority as the clock.

And that word already hints at what would probably make more sense:
Not just a boolean but a list of priorities, probably.

This makes the current FR here obsolete and given that I already looked at the code, it makes _any_ FR for this obsolete, because if I should come up with an idea for that, I might as well just implement it.

So thanks a lot for the software and the input!

closed 2025-04-11 by Hypfer