Blueforcer/awtrix3 #724: VOL parameter shows "NA" on display when trying to set volume

rmribeiro2024, 2025-04-13:
The VOL parameter doesn't seem to work as expected.

When I try to change the volume via MQTT through Home Assistant, nothing happens. The volume remains unchanged and there's no feedback on the display.

When I try to change the volume physically using the device button, the screen shows "NA" and no adjustment is made.

Example of the MQTT payload via Home Assistant:

service: mqtt.publish
data:
  topic: AWTRIX/notify
  payload: >
    {
      "text": "Volume test",
      "VOL": 20,
      "sound": "alarm",
      "color": [255, 255, 255]
    }

Expected behavior:

The buzzer or DFPlayer volume should change according to the value provided in the VOL parameter.

The display should not show "NA".

The physical button should allow volume adjustment directly on the device.

Obs.:  
1) The text message shows normally.
2) By app awtrix 3 android app I can't change vol too. It doesnt work.

Additional info:

Firmware version: v0.96

Device: ulanzi tc001

Tested with: MQTT / HOME ASSISTANT

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [096

Blueforcer, 2025-04-13:
The Ulanzi doesn't have a Dfplayer. Only a buzzer.
Changing a buzzers volume is a dirty hack, because physically it can't work on this device.
That's why you need to enable the volume control of the buzzer via dev.json.
I don't recommend it, because some tones doesn't works and ends up with a continuous tone.

rmribeiro2024, 2025-04-13:
> The Ulanzi doesn't have a Dfplayer. Only a buzzer. Changing a buzzers volume is a dirty hack, because physically it can't work on this device. That's why you need to enable the volume control of the buzzer via dev.json. I don't recommend it, because some tones doesn't work and ends up in a continuous tone.
> 
> Also you can't mix /settings API and /notify commands. You need to send them to their own endpoints

Thank you for the clarification and for explaining the correct use of the /settings and /notify endpoints.

I understand that the Ulanzi only has a buzzer and not a DFPlayer. I wasn’t trying to play complex sounds — I just needed the Ulanzi to emit a small beep, but with a lower volume, since the current one is quite loud and ends up being a bit annoying in my environment. If it were possible to get a small beep, I would be happy with that. Do you think it's not feasible?

Thanks again for your help and the guidance on using the correct API endpoint!

Blueforcer, 2025-04-13:
Try to enable volume buzzer control with the dev.json (hidden features in the docs)
But I can't give any promises. If this doesn't fit your needs, I can't help anymore. Sorry

rmribeiro2024, 2025-04-13:
thank you very much for your support. I’ve made the changes you suggested by enabling "buzzer_volume": true in the dev.json, and after rebooting, I can now see the “Sound: ON” indication on the AWTRIX display.

However, when I try to send a volume setting via MQTT right before triggering a sound, the volume doesn’t seem to change. Here’s an example of the code I’m using:

- action: mqtt.publish
  data:
    topic: AWTRIX/settings
    payload: |
      {
        "VOL": 2
      }

- action: mqtt.publish
  data:
    topic: AWTRIX/notify
    payload: >-
      {
        "text":"TEST SOUND",
        "duration": 10,
        "icon":"Love_423",
        "sound": "chime",
        "color": [255, 255, 255]
      }
I’ve tested different sounds, but the volume always sounds the same.

Is there anything else I should check to make sure the volume control is applied correctly? 

Thanks again for your help and for this great project!