Blueforcer/awtrix3 #728: [BUG] Animated images after firmware update to V0.98 cause system crash

UrsKobler, 2025-05-13:
# Bug report

## Describe the bug

If apps display animated images in addition to text, the system may crash, depending on the number of animated images used.

Afterward, the display can only be restarted from the frozen state by manually resetting it.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: [v0.98 ]
  - used interface: HTTP API

- My System requirements:
  - Platform Operating system: Linux
  - Architecture: Arm64
  - CPUs: 4
  - Speed: 2400 MHz
  - Model: Cortex-A76
  - RAM: 7.8 GB
  - Node.js: v20.19.1
  - NPM: 10.8.2

## To Reproduce

Steps to reproduce the behavior:

1. Use the HTTP API interface of the Ulanzi Awtrix Smart Pixel Clock.

2. Create App-1 with the text 'App-1' and the animated image ID 50065.

3. Create App-2 with the text 'App-2' and the animated image ID 50556.

3. Create App-3 with the text 'App-3' and the animated image ID 51008.

4. Create App-4 with the text 'App-4' and the animated image ID 55417.

5. Create App-5 with the text 'App-5' and the animated image ID 61524.

6. Create App-6 with the text 'App-6' and the animated image ID 7789.

7. Display the apps on the screen and wait approximately 1 hour until the screen freezes.

## Expected behavior

The display remains frozen and can only be reset by manually restarting.

## Screenshots

[image]

## Logs

The firmware crashes so no log file can be written.

## Additional context

I suspect the error lies in the program component responsible for displaying the animated images (GIF files). Presumably, a memory leak occurs every time the image is displayed, which consumes RAM and causes the program to crash.

nob0dy80, 2025-05-23:
Hey Urs,

I’ve now tried to reproduce your issue on my test-Ulanzi.
I created 6 apps using the icons you mentioned (even in the same order).

No problems at all. It’s been running for 24 hours now, so I think we’ve clearly passed the “approximately one hour” mark. I don’t think it’s a firmware issue.

What have you already tried to solve the problem?

UrsKobler, 2025-05-23:
Hey BlueForcer,

Thank you for running the test I described to reproduce the phenomenon.

As already described, the system usually freezes after 48 hours of operation.

The problem can probably be reproduced more quickly by setting the cadence for displaying the animated GIF files as high as possible.

However, since the error usually reproducibly occurred after 48 hours in my described test environment, I assumed that other users would also experience it after 48 hours at the latest!

Since I am certain that the error reproducibly occurs after at least 48 hours, I have only operated the display with pure text output, as the display plays an important role in my production system for monitoring the energy flows of my PV solar system.

I would be grateful if you could continue the test, as I am sure that after 48 hours the display will freeze.

nob0dy80, 2025-05-23:
Hi Urs,

Unfortunately, you mentioned two different time spans in your error report, which you just corrected.
So we're not talking about roughly an hour, but actually about 48 hours.
It would’ve been nice if you had said you made a mistake in the original report.

But no worries—I've got a few of these devices.
I'll let it run for another 1–2 days and we’ll see what happens.

[image]

UrsKobler, 2025-05-23:
Hello nob0dy80,

Please excuse my mistake with the incorrect test duration. I only realized after your message that I had made a mistake with the test duration of 1 hour.

Since the values ​​displayed on my display depend partly on the outside temperature, wind speed, and whether it's raining, the duration always varied slightly.

But after 24 hours at the latest, I was forced to manually restart the display.

[image]

nob0dy80, 2025-05-23:
What you're doing with it can largely be inferred from the icons.

I really like your wall mount for it :-) And above all, it seems like you've somehow made the power supply invisible. Very nice. Definitely a benchmark :-)

As I said, I'll just let it run for now, but I have a strong suspicion that the issue isn't with the firmware itself.

Blueforcer, 2025-05-23:
I doubt that as well. Awtrix now has a user base in the five-digit range. Having 6 apps with icons is nothing unusual otherwise I would have lots of issues this kind. I myself have 15 apps running 24/7 without any issues. I suspect it must be something else. Maybe you're using other API commands too?

nob0dy80, 2025-05-23:
@Blueforcer Well, nothing's impossible... I'm thinking it might be caused by a faulty GIF icon. So I'll just let it run and see what happens.
Since @UrsKobler didn’t mention what he tried to fix it, I’d generally recommend reflashing AWTRIX.
But let’s see what time reveals...

UrsKobler, 2025-05-23:
If the error only occurs on my device and I'm the only one experiencing this strange phenomenon, I can only imagine that the cause could be a corrupted GIF file that may have been damaged during upload to the display.

If you haven't been able to reproduce the error until tomorrow, I'll try deleting all GIF files on the display and then reloading them.

UrsKobler, 2025-05-24:
In the meantime, I deleted all GIF files and then reloaded them onto the display. I then ran the test with the animated GIF files overnight.

[image]

Unfortunately, this morning I discovered that the display had frozen again.

The awtrix-light FreeRAM showed the following values:

**Test start 05/23/2025-8:36 PM:**    113'796 bytes

**Test end 05/24/2025-8:37 AM:**     119,176 bytes

If I'm the only user in the world experiencing this phenomenon, I have to assume that something is malfunctioning with the hardware.

I could imagine that a thermal effect in the charging electronics could lead to the described phenomenon under higher loads due to the GIF images displayed.

Since the animated images are purely cosmetic, I'm fine with the purely textual information on the display, so I won't be buying a new display at the moment as long as it's still usable this way.

Blueforcer, 2025-05-24:
Did you tried my other recommendations?
And are you using a unifi WiFi?

UrsKobler, 2025-05-24:
Hi Blueforcer,

I use the WiFi Mesh System from the German manufacturer AVM. It contains the Fritzbox 5530 AON router and several AVM access points connected via LAN.

I've disabled the internal apps (temperature, clock, etc.) since the beginning, and I don't use any custom apps either.

Blueforcer, 2025-05-24:
II meant if you use any other APIs besides customapps.

UrsKobler, 2025-05-26:
Hello nob0dy80,

First of all, I would like to thank you for your support in determining the cause of the error with my display.

As a final step, I will gladly follow your recommendation and reflash firmware version 0.98.

I will be happy to report on the outcome of the final long-term test.

Best regards,
Urs

warp735, 2025-06-03:
Unfortunately, I have exactly the problem described here with two Ulanzi's
My displays freeze every day. Sometimes after restarting after only 5 minutes

I reported this here a year ago and am always amazed that so few people seem to be affected.

Since the problem has existed for a very long time and there is no improvement in sight, I will now do without animated GIFs and reprogram my display.

Blueforcer, 2025-06-03:
As long as the problem is not reproducible, but only occurs with 2-3 users out of over 10k, I unfortunately have to assume a hardware defect.
I hope you understand that my hands are tied here. I can't fix an avoidable bug that doesn't exist.

warp735, 2025-06-03:
If I only had one device, I might also consider a defect. But with two devices, that's unlikely. What's more, they work wonderfully with static images.

I also use the ioBroker adapter. But in the stable version 1.7.0

Incidentally, I have had this problem since all versions. So it's not specifically due to 0.98

UrsKobler, 2025-06-04:
@Blueforcer

Unfortunately, I was too quick to celebrate the results of my first hardware reset.

I just noticed that the display has frozen again after about 17 hours.

So, this unpleasant phenomenon persists, and I have to, for better or for worse, get to grips with the esptool for creating the image dump, which is new to me. If I have any problems with it, I would be grateful if you could support me with your experiences.

Blueforcer, 2025-06-04:
@UrsKobler
Thats not a "hardware" reset button. It's a regular button wich is read by the ESP to do any kind of software stuff. In this case clearing the flash and settings etc which is also done during a reflash.

To be clear.. You're not using animated icons and its still freezing?

eku, 2025-06-05:
@UrsKobler you may want to try out #564