Blueforcer/awtrix3 #729: [FEATURE REQUEST] Wake on Power / Wake On Lan functionality

delize, 2025-05-20:
# Feature Request

### Is your feature request related to a problem?

I turn off power to my desk when I sleep, and so when I wake up, the Awtrix is likely dead at the beginning of the day and I have to manually turn it on. 

Even if I put the display to sleep, this still becomes a problem, as it doesn't appear to keep a charge long enough during a 6 - 12 hour sleeping period.

### Describe the solution / feature you'd like

When power is detected on the ESP32 board (specifically in my case, Ulanza TC0001), allow for the ability to automatically power on the device or wake the device up if it's asleep. Similar to a Wake On Lan, but, Wake On Power (or something similar). 

I already have an automation to turn the device's pixel screen on when I get out of bed through Home Assistant, but because the issue is the charge held or battery level, a screen on automation won't work in all cases.

### Describe alternatives you've considered

The only other option I see here, is rewiring my whole desk and buying a ton of additional power smart plugs so that the majority of other things can be plugged into smart adapters and this can be left on all the time. However, trying to save electricity where ever possible. 

### Additional context

None

Blueforcer, 2025-05-20:
That's not possible.. When the clock is off, the ESP32 has no power to operate.

closed 2025-05-20 by Blueforcer

delize, 2025-05-20:
Bummer, thanks for the response in any case. :(