Blueforcer/awtrix3 #740: [BUG] Time is behind several minutes

Rinyuu, 2025-06-27:
Minor issue but it started to irritate me after running it for months; time is always consistently late about two minutes. (Compared with all of my other devices, PC's, Smartwatch, phones, consoles etc.) 

I tried reflashing, re-setting the time server and timezone, using different timeservers and obviously rebooting the device many times, but all to no avail. It seems like a pretty persistent issue... 

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.98

As far as I could find, there is no way to manually offset the time in Awtrix?

Blueforcer, 2025-06-27:
Awtrix gets it's time from a NTP server, there is no offset.
Is the time still wrong after a reboot?

adnet-solutions, 2025-07-05:
i have the same issue. i have 2 awtrix here. both running version 0.98
one show the right software, the second one have same settings but shows a complete wrong time and date (date is 01.01. and time something like 01:11 after restart. Looks like it not get time from timeserver, but its connectd to wifi and internet (iobrioker connection works)

Blueforcer, 2025-07-05:
> i have the same issue. i have 2 awtrix here. both running version 0.98
> one show the right software, the second one have same settings but shows a complete wrong time and date (date is 01.01. and [...]

So it's not the same issue.
Iobroker connection doesn't proof that it has internet. Make sure you're router doesn't block, or any custom DNS like pihole, ad guard etc set it on the blacklist. If you're using a static IP make sure you set the gateway correct

adnet-solutions, 2025-07-05:
> So it's not the same issue. Iobroker connection doesn't proof that it has internet. Make sure you're router doesn't block, or any custom DNS like pihole, ad guard etc set it on the blacklist. If [...]

I double check that allready and i also not have any pihole or something like that running which can block the access. Is there any way on the awtrix to check the internet connection or is there any log on the system ro see if he is blocked?

adnet-solutions, 2025-07-05:
> That does sound like it's unrelated to my issue. My time never is completely off, it's always NEARLY correct, just a few minutes early.
> 
> I also would like to emphasize that in my case Awtrix is correctly connected to the web, since changing the timezone or timeserver works as it should and updates correctly (at least after a reboot). It's just that it's always about 2 minutes late, regardless of timezone / reboots / reflashes etc. compared to all my other devices.

@Blueforcer shall i raise a separate ticket?

Rinyuu, 2025-07-12:
> Try this solution: [#716 (comment)](https://github.com/Blueforcer/awtrix3/issues/716#issuecomment-2821198621)

Thanks! Great tip, but unfortunately it didn't change or fix anything. I tried a few of the different IP's pool.ntp.org resolves to. Still about two minutes behind.