Blueforcer/awtrix3 #743: [BUG] Calendar and todolist won't update

PaulBiod, 2025-07-14:
Hi, I'm using the  @smarthomejunkie  blueprints, so not sure if it's awtrix or blueprint related.

If I use the todolist items blueprint, it works, but adding or removing items from the list won't update the ulanzi screen.

Same behavior for calendar. If I create a calendar, and an event occurs, the screen show the event, but next day, when the event finished, the screen still shows the event.

If i switch off and back on the toogle helper for calendar / todolist, then it updates successfully.

I tried to go to the discord channel, but even if I am a Ko-fi member, it seems I am not allowed to post.
So, if @smarthomejunkie is around, any help would be appreciated. Thanks

PaulBiod, 2025-07-14:
@smarthomejunkie 
Thanks, I will send you a mesg through Ko-fi.
I made a small video to show you.
I updated with your last v3.03, and just created the todolist items automation from blueprint.
For the video, I just edited the blueprint to update every 10 seconds, not to wait 1 min for the update.
The link to the video : 

https://youtu.be/96vZeroAJFI

Thanks again

webash, 2025-07-24:
This doesn't seem like an issue with the firmware.

I'd suggest opening a MQTT client and watch the Awtrix's topic to see if the right mqtt messages are sent when you expect them to. I'd also try sending manual mqtt messages to the Awtrix topic crafted by hand to see if you can make it happen as you expect.

If all looks fine there, then there might be something wrong between Home Assistant and the MQTT broker - not Awtrix.

smarthomejunkie, 2025-07-24:
Just uploaded a new version to: https://ko-fi.com/s/0d1e4419bd. The todo blueprint is completely refactored. Thanks for noticing @PaulBiod 

This issue can be closed.

closed 2025-07-25 by PaulBiod