Blueforcer/awtrix3 #744: [BUG] "Current App" in Home Assistant does not update when clicking "Next app" and "Previous app" (but does when automatically transitions)

octane3003, 2025-07-20:
# Bug report

## Describe the bug

"Current App" in Home Assistant MQTT Integration does not update when clicking "Next app" and "Previous app" (but does when apps automatically transition)

## Additional information

- Devices involved:
  - Model: ESP32 with generic panel
  - awtrix3 version: 0.98

## To Reproduce

Steps to reproduce the behavior:
1) Create several apps with automatic transition
2) Click "Next app" or "Previous app" in the MQTT integration

## Expected behavior

"Current App" in the MQTT integration should update the currently displayed app that was manually transitioned to. 

## Actual behaviour

"Current App" in the MQTT integration stays at the previously displayed app until it automatically transitions again and then it updates to the correct app. 

## Thanks for the great firmware

webash, 2025-07-24:
This works for me on 0.98 and HA 2025.6.3 running the vanilla Moquitto MQTT server as a docker container.

Can you open an MQTT message client and subscribe to the current app topic to see what information is returned directly to make sure it isn't a Home Assistant or MQTT fault?

octane3003, 2025-12-15:
I reinstalled on a new esp32 and could not reproduce the behaviour. You can close this issue.

Minor related issue is that when I delete the current app, the current app in HA does not update until after an automatic transition.