Blueforcer/awtrix3 #746: [BUG] power API turns off matrix, but cannot enable it after

thoastbrot, 2025-08-02:
# Bug report

## Describe the bug

Similar to #650 I cannot reenable the matrix after turning it off. state/matrix in mqtt reports true/false correctly, but the display stays off.

To switch, I used these (no wrong quoting here):

`curl -d '{"power":false}' http://192.168.1.54/api/power`
`curl -d '{"power":true}' http://192.168.1.54/api/power`

With variants like "on"/"off" (which also were accepted, sadly there's no GET API)

Afterwards, I have to reset them via buttons. A reset API would be nice.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: 0.98 (flashed via web)

Blueforcer, 2025-08-02:
I wrote the solution in the issue you linked:
https://github.com/Blueforcer/awtrix3/issues/650#issuecomment-2560317179

Blueforcer, 2025-08-02:
Escaping depends on your terminal.
Does my example works or not?