Blueforcer/awtrix3 #747: [BUG] Matrix freezes for 10 seconds every minute when dfplayer option enabled

filippcanadian, 2025-08-03:
# Bug report

## Describe the bug

Matrix freezes for 10 seconds every minute when dfplayer option enabled.
LiveView tab also freezes
Issue also replies with no DFPlayer module connected.
Once your delete "dfplayer": true from dev.json, freezes gone

dev.json config:

{
  "matrix": 2,
  "ap_timeout": 120,
  "dfplayer": true
}

## Additional information

- Devices involved:
  - Model: Custom Awtrix installation with ESP32 DevKit V4
  - awtrix3 version:  ulanzi_TC001_0.98.bin 

## To Reproduce

Add   "dfplayer": true to dev.json

Blueforcer, 2025-08-03:
Does your dfplayer work with awtrix?

filippcanadian, 2025-08-03:
Yes, my dfplayer and speaker work fine with awtrix in this installation

closed 2025-08-05 by filippcanadian