Blueforcer/awtrix3 #757: [BUG] Parameter "pos" not working

cbornemann, 2025-09-16:
# Bug report

## Describe the bug

When i push a new app with the parameter "pos", than the parameter is ignored and the position of the new app is always on the end of the loop.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - awtrix3 version: v0.98

## To Reproduce

Push a new app with the parameter "pos".

## Expected behavior

When i push a new app with pos=0 after an app with pos=1, the new app is on position 1 now.

Blueforcer, 2025-09-16:
According to the documenation, this function is still experimental. It is expected to be removed in future versions.

Blueforcer, 2025-09-16:
Send them in the desired order, or you can try to use a hidden and also experimental api /reorder by sending a json array with the apps name after all apps are sent. Like
["time","date","custom1"]

Blueforcer, 2025-09-16:
Thats correct. Apps are saved temporarily in the RAM. Because it take care of the flash wroting cycles and in normal operation your smarthome sends new app anyways at each sensor update. You can listen to the awtrix online state and send the reorder after it gets online or after all apps sent their first states

casdevs, 2025-10-03:
> Send them in the desired order, or you can try to use a hidden and also experimental api /reorder by sending a json array with the apps name after all apps are sent. Like ["time","date","custom1"]

@Blueforcer: would it be possible to also enable the /reorder API via MQTT in the next release? At the moment, this only works via HTTP...